ARMmbed/mbed-os #9737: CellularStateMachine can generate spurious attach events

NeilMacMullen, 2019-02-15:
### Description
In the following code...
```
         UARTSerial * serial = new UARTSerial(....);
	UBLOX_AT * dev = new UBLOX_AT(serial);
	CellularContext* iface = dev->create_context(serial, "stream.co.uk", false, false);
	iface->set_credentials(..., ..., ...);
	iface->connect();
	dev->set_timeout(5000);
```

The intended timeout of 5000 can be overwritten and reset to 60000.  This appears to be because the CellularStateMachine is driven by a separate thread and so the "connect" call can return before all events are processed.  Specifically, the "state_attaching" can be called _twice_, setting the timeout both times.  This appears to be because it uses the current _network_status as a guard but fails to set it.  The following version of the code behaves a bit better though I'm not sure if it completely addresses the issue.

```
void CellularStateMachine::state_attaching()
{
	if (_network_status != ATTACHED_TO_NETWORK) {
	    _cellularDevice.set_timeout(TIMEOUT_CONNECT); //only set the timeout if we are trying to attach
        _cb_data.error = _network->set_attach();
       }
       if (_cb_data.error == NSAPI_ERROR_OK) {
		_network_status |= ATTACHED_TO_NETWORK; //ensure we know we're attached.
                if (_event_status_cb) {
                    _cb_data.status_data = CellularNetwork::Attached;
                    _event_status_cb(_current_event, (intptr_t)&_cb_data);
                }
      } else {
        retry_state_or_fail();
     }
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jarvte, 2019-04-04:
This issue will be fixed in this pr: https://github.com/ARMmbed/mbed-os/pull/10053

jarvte, 2019-05-13:
More fixes in https://github.com/ARMmbed/mbed-os/pull/10511

closed 2019-05-23 by ciarmcom