ARMmbed/mbed-os #9739: Mbed OS tests fail to build with IAR 7.8 + CY8CKIT_062_WIFI_BT + Mbed OS 5.11.5-alpha

maclobdell, 2019-02-15:
### Description
Mbed OS Version: 5.11.5-alpha
Target: CY8CKIT_062_WIFI_BT 
Compiler Version: IAR 7.8

I'm starting with a fresh when I issue the test command to launch tests
 ``` mbed test -m CY8CKIT_062_WIFI_BT -t IAR --compile -vv```

```
        [Error] objects.h@80,6: [Pe020]: identifier "cy_en_divider_types_t" is undefined
        [Error] objects.h@112,6: [Pe020]: identifier "cy_en_divider_types_t" is undefined
        [Error] cy_ctdac.h@561,0: [Pe020]: identifier "cy_en_divider_types_t" is undefined
        [DEBUG] Return: 2
        [DEBUG] Output:
        [DEBUG] Output:       cy_en_divider_types_t       div_type;
        [DEBUG] Output:       ^
        [DEBUG] Output: "C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR\targets\TARGET_Cypress\TARGET_PSOC6\objects.h",80  Error[Pe020]: identifier "cy_en_divider_types_t" is undefined
        [DEBUG] Output:
        [DEBUG] Output:       cy_en_divider_types_t               div_type;
        [DEBUG] Output:       ^
        [DEBUG] Output: "C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR\targets\TARGET_Cypress\TARGET_PSOC6\objects.h",112  Error[Pe020]: identifier "cy_en_divider_types_t" is undefined
        [DEBUG] Output:
        [DEBUG] Output:       cy_en_divider_types_t       dividerType;        /**< Specifies which type of divider to use. Can be integer or fractional divider. Not used if updateMode is \ref CY_CTDAC_UPDATE_DIRECT_WRITE */
        [DEBUG] Output:       ^
        [DEBUG] Output: "C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR\targets\TARGET_Cypress\TARGET_PSOC6\psoc6pdl\drivers\include\cy_ctdac.h",561  Error[Pe020]: identifier "cy_en_divider_types_t" is undefined

[mbed-47632] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy"
       Command: "c:\python27\python.exe -u C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy\tools\test.py -t IAR -m CY8CKIT_062_WIFI_BT --source . --build .\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR --test-spec .\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR\test_spec.json --build-data .\BUILD\tests\CY8CKIT_062_WIFI_BT\IAR\build_data.json -v --greentea"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```
Please note that building a simple blinky program works fine.  

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-02-15:
@lrusinowicz @vmedcy Please review

0xc0170, 2019-02-15:
Master builds? Is it only release 5.11.5 and IAR affected?

cmonr, 2019-02-15:
@maclobdell What test was it trying to compile? Looks like that was cut off...

vmedcy, 2019-02-15:
I cannot reproduce this in mbed-os-5.11.5-alpha tag.
targets/TARGET_Cypress/TARGET_PSOC6/objects.h includes cy_sysclk.h.
The sysclk header provides cy_en_divider_types_t unconditionally.

https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.11.5-alpha/targets/TARGET_Cypress/TARGET_PSOC6/objects.h#L30
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.11.5-alpha/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_sysclk.h#L1844
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.11.5-alpha/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_ctdac.h#L373

Hence the error `identifier "cy_en_divider_types_t" is undefined` cannot happen if the header include search paths are correct.

@maclobdell: can you attach the complete build log, and confirm there is no custom mbed_app.json.

maclobdell, 2019-02-15:
here is the full log
[mbed-os-tests-5.11.5-alpha_iar.txt](https://github.com/ARMmbed/mbed-os/files/2870138/mbed-os-tests-5.11.5-alpha_iar.txt)

cmonr, 2019-02-15:
@maclobdell The log you uploaded, are you aware it's trunkated?

It doesn't contain the command run, or the first couple of lines.

vmedcy, 2019-02-15:
@maclobdell:
I checked this sequence:
cd mbed-os
git checkout mbed-os-5.11.5-alpha
git clean -xdf
mbed test -t IAR -m CY8CKIT_062_WIFI_BT -n tests-mbed_platform-system_reset --compile

It doesn't produce any errors.
I do not see how objects.h can include cy_sysclk.h without cy_en_divider_types_t - maybe there is another cy_sysclk.h in the resources (outside of the git index) with higher header include search priority?

maclobdell, 2019-02-15:
@cmonr  - yes, aware it is truncated.  

here is the sequence I used

mbed import mbed-os mbed-os-cy    <- this clones the mbed-os repo into directory 'mbed-os-cy', checks out the latest release branch.
cd mbed-os-cy
mbed test -m CY8CKIT_062_WIFI_BT -t IAR --compile -vv

cmonr, 2019-02-15:
@maclobdell There's your problem.

Due to the weirdness of the `mbed-os-5.11.5-alpha`, the `latest` tag still points to the 5.11.4 release.

I'm about to finish locally testing, but I'm not finding any problems. Similar string of commands as @vmedcy .

@ARMmbed/mbed-os-maintainers Should the `latest` tag be updated as well?

maclobdell, 2019-02-15:
i'm checking if this is due to a caching problem.  I've explicitly disabled caching but it still seems to be pulling my mbed-os version from a local place

cmonr, 2019-02-15:
Talking with @adbridge, since 5.11.5-alpha isn't an "official" release per-se (no release notes), the `latest` tag was not updated on purpose.

maclobdell, 2019-02-15:
This issue seems to be resolved.  I'm building successfully now. 

I'm fairly certain that this issue is due to a corrupted version of mbed-os on my machine.  When I was importing I was using the shorthand 'mbed import mbed-os', instead of the full URL.  Even though I was pulling the latest updates and changing to the correct branch, it seems that there were missing files.

closed 2019-02-15 by cmonr