ARMmbed/mbed-os #9744: Hard Fault when trying to call UDPSocket::join_multicast_group

pauluap, 2019-02-15:
### Description

Target: STM32F769-Discovery evaluation board
Toolchain: GCC_ARM (C++11)
mbed-os: v5.8.3, sha: c05d72c

Repository containing failing code: https://github.com/pauluap/mbed-os/tree/UdpMulticastIssue

I'm trying to set up UDP multicast (both send and receive) and am running into problems when trying to call `UDPSocket::join_multicast_group()`

I have replicated the issue in a minimal test case at the repository branch linked above.

When I try to run the application, I get a hard fault; when I use mbed-os' `crash_log_parser.py` to parse the fault, I get this output:

```
Crash Info:
	Crash location = Socket::modify_multicast_group(SocketAddress const&, nsapi_socket_option) [0x08015516] (based on PC value)
	Caller location = Socket::modify_multicast_group(SocketAddress const&, nsapi_socket_option) [0x080154FF] (based on LR value)
	Stack Pointer at the time of crash = [C0002CF8]
	Target and Fault Info:
		Processor Arch: ARM-V7M or above
		Processor Variant: C27
		Forced exception, a fault with configurable priority has been escalated to HardFault
		Unaligned access error has occurred
```

Can you provide any guidance in resolving this issue?

Thank you.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

pauluap, 2019-02-16:
I removed the `sbrk` implementation from my original commit because the hard fault wasn't the issue that I was experiencing in my real hardware, the hard fault was an distraction.

With `sbrk` removed, I now experience the same issue as in my actual hardware - the problem is that when executing `join_multicast_group`, when the code reaches `do_memp_malloc_pool` in `lwip_memp.c`, it reaches line 304, `SYS_ARCH_PROTECT(old_level);` the code never reaches the next line.  I take this to be a deadlock somehow.

Oddly, when I put a breakpoint in `igmp_lookup_group` in `lwip_igmp.c` - which is part of the `join_multicast_group` call stack, I see it being called before my problematic code is reached.  I assume that it's part of the LWIP startup process.  So in one case, group creation works, in the other (my use case), it doesn't.

Sorry for the distraction about the hard fault, this is the issue I actually want resolved.

jeromecoutant, 2019-02-20:
> @naveenkaje added [devices: st]

Not so sure issue is on ST side ?
I think you should assign the issue to @mbed-os-ipcore ?

kjbracey-arm, 2019-02-20:
It looks like it may be an lwIP porting bug. Comments in our porting layer say:

> This function should support recursive calls from the same task or interrupt. In other words, sys_arch_protect() could be called while already protected.

But we're using a non-recursive CMSIS-RTOS mutex, as far as I can see. Can you try adding 

    lwip_sys_mutex_attr.attr_bits = osMutexPrioInherit | osMutexRecursive;

to `sys_init` in `lwip_sys_arch.c`?

(The Mbed OS `Mutex` C++ wrapper always uses those flags).

pauluap, 2019-02-20:
Thank you.  Making the modification you suggested does indeed allow the multicast group join call to complete.

I have a followup question.  I tried compiling the toy example with mbed-os 5.11.4, and the issue doesn't manifest itself.  I'm working on trailing an upgrade to 5.11.4 for my application and saw that the modification isn't in the 5.11.4 source. 

My question is - if moving to 5.11.4 ends up working out for me, should I still go ahead and perform the modification?  Or has the apparent recursive locking scenario been circumvented in another way?

jeromecoutant, 2019-06-17:
@pauluap 
Do you plan to push a pull request ?
Should we close this issue ?
Thx

pauluap, 2019-06-17:
Well, the change does work for mbed 5.8.3.  I can make a pull request for that if it turns out to be useful.

I still don't know why the issue seems to be less likely on mbed 5.11.  I *THINK* that it's still present, I did encounter a scenario that seems suspiciously like the deadlock described in this issue, but it happened so rarely, I wasn't able to confirm.  I decided to just roll with using non-blocking UDP sockets as my workaround.

I'd be hesitant about requesting this change as a pull for master because I don't understand things well enough to know what help or harm it'll do especially since I don't know what changed between 5.8 and 5.11 to make the deadlock incidents less likely to occur.

kjbracey-arm, 2019-06-18:
It is a clear lwIP porting bug - if that mutex is still not recursive, it still needs to be fixed. I don't recall making a PR myself - I think I may have been expecting you to...

pauluap, 2019-06-18:
Allright, I created a pull request against master.  I am currently based on mbed-os 5.11.3 with nonblocking sockets, so I don't run in this issue myself anymore.  The best I can do is performing some non-exhaustive stress testing with several TCP and UDP streams.  My system hasn't died so far.

The only deterministic test case I'm aware of is the original one I created with mbed-os 5.8

PR #10861 (2019-07-01): Make the sys_arch_protect() mutex be recursive.
### Description
Fixes #9744

Appears to be a lwIP porting bug. Comments in our porting layer say:

    This function should support recursive calls from the same task or interrupt. In other words, sys_arch_protect() could be called while already protected.

### Pull request type

    [x ] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
commit 91402f4d (2019-07-01): Make the sys_arch_protect() mutex be recursive.

Fixes #9744

Appears to be a lwIP porting bug. Comments in our porting layer say:

    This function should support recursive calls from the same task or interrupt. In other words, sys_arch_protect() could be called while already protected.
closed 2019-07-01 by 0xc0170