ARMmbed/mbed-os #9750: Bootloader merging fails with python3, [ERROR] 'l' format requires -2147483648 <= number <= 2147483647

teetak01, 2019-02-19:
### Description

Severity: Blocker (breaks build), cannot work in CI-automation.

Mbed OS master #31da50e8fa68a2bbe8c92d70b258a5d19a41b5e3 (19.02.2019)
mbed-cli 1.9.1
Python3 + K64F + ARMC6
mbed-cloud-client-example (internal development version)

Bootloader merging fails with:
```
Merging Regions
  Filling region bootloader with /home/teetak01/devel/mbed-cloud-client-example-internal/tools/mbed-bootloader-k64f-block_device.bin
[ERROR] 'l' format requires -2147483648 <= number <= 2147483647
[mbed] ERROR: "/usr/bin/python3" returned error.
```

This fails two times, and third retry in compilation passes.

```
Merging Regions
  Filling region bootloader with /home/teetak01/devel/mbed-cloud-client-example-internal/tools/mbed-bootloader-k64f-block_device.bin
  Filling region header with ./BUILD/K64F/ARMC6/mbed-cloud-client-example-internal_header.hex
  Filling region application with ./BUILD/K64F/ARMC6/mbed-cloud-client-example-internal_application.bin
Space used after regions merged: 0x6df34
Merging Regions
  Filling region application with ./BUILD/K64F/ARMC6/mbed-cloud-client-example-internal_application.bin
Space used after regions merged: 0x63b34
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-02-19:
There was this fix but caused some other failures https://github.com/ARMmbed/mbed-os/pull/8788, reverted via https://github.com/ARMmbed/mbed-os/pull/8915

cc @ARMmbed/mbed-os-tools

teetak01, 2019-02-19:
I tested this with Mbed OS from last friday, and this morning.

bridadan, 2019-02-19:
Hi @teetak01 I wasn't able to reproduce this, here's my environment:

Windows 10
Mbed CLI 1.9.1, installed on python 3.6.3
Hash for `mbed-cloud-client-example-internal`: 991357060e45569393ded95d6dc37b4f438df3f2
Hash for `mbed-os`: 31da50e8fa68a2bbe8c92d70b258a5d19a41b5e3
Command: `mbed compile -m K64F -t ARMC6`

Do you have a setup that reproduces the issue consistently?

cmonr, 2019-02-19:
@bridadan Sanity check. Could you retry the previous command with `-v` and `-vv`?

Iirc, I came across this problem last week (was retesting this: https://github.com/ARMmbed/mbed-os/issues/8146#issue-360754864), and this odd behavior led me to opening https://github.com/ARMmbed/mbed-os/issues/9675

theotherjimmy, 2019-02-19:
@cmonr That was related to the time stamp, so running with `-v` was eliminated as the cause.

bridadan, 2019-02-19:
(for completeness I ran it with `-v` and `-vv`, no change)

teetak01, 2019-02-20:
@bridadan Ubuntu 18.04.1, Python 3.6.7, fresh installation.

teetak01, 2019-02-20:
Also happens with GCC 6.3.1

bridadan, 2019-02-20:
I had a spare moment to setup an Ubuntu VM. I am able to reproduce the error!

bridadan, 2019-02-20:
This one was a quick fix once I had that reproducible case. Here's the PR: https://github.com/ARMmbed/mbed-os/pull/9783/files

PR #9783 (2019-02-21): Ensure header CRC is written as unsigned int
### Description

Fixes #9750. The bootloader merge would fail since the CRC produces an [unsigned 32 bit integer](https://docs.python.org/3/library/zlib.html#zlib.crc32), however we specified it as a [signed 32 bit integer](https://docs.python.org/2/library/struct.html#format-characters).

I've tested this on a K64F and ran an update over Pelion, seems to work ok!

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@the [... cut]
closed 2019-02-21 by cmonr