ARMmbed/mbed-os #9757: USBSerial vendor_id=0x1f00, product_id=0x2012. Whose is the default value of this VID and PID?

maskedw, 2019-02-19:
### Description

`features/unsupported/USBDevice/USBSerial/USBSerial.h`

```c++
    /**
    *   Constructor
    *
    * @param vendor_id Your vendor_id (default: 0x1f00)
    * @param product_id Your product_id (default: 0x2012)
    * @param product_release Your preoduct_release (default: 0x0001)
    * @param connect_blocking define if the connection must be blocked if USB not plugged in
    *
    */
    USBSerial(uint16_t vendor_id = 0x1f00, uint16_t product_id = 0x2012, uint16_t product_release = 0x0001, bool connect_blocking = true): USBCDC(vendor_id, product_id, product_release, connect_blocking){
        settingsChangedCallback = 0;
    };
```

vendor_id: 0x1f00, product_id: 0x2012

Whose is the default value of this VID and PID?
Am I allowed to use this default value for my device?

### Issue request type

[x] Question
[ ] Enhancement
[ ] Bug

cmonr, 2019-02-19:
@c1728p9 Would you know?

c1728p9, 2019-02-19:
I don't know what the above VID and PID correspond to. If your company has a VID then I would recommend using it rather than the mbed-os default.

cmonr, 2019-02-19:
>  Am I allowed to use this default value for my device?

@maskedw For prototyping, sure. But once you go into production and start selling your device, you'll need to either get your own VID/PID combination, or use one provided by a partner/vendor.

maskedw, 2019-02-21:
understood. Thank you for answering.

closed 2019-02-21 by maskedw