ARMmbed/mbed-os #9760: Cellular connectionf fails on sequential connect

vidavidorra, 2019-02-19:
### Description

mbed-cli: 1.8.3
toolchain: cmake_gcc_arm
gcc-arm-none-eabi: 7-2018-q2-update (7.3.1)
mbed-os: 5.11.1

I'm using the CellularContext as a cellular interface and am running into the issue where I don't get more than one connect working. The connect only works once when I reset the device.

I'm using a non-blocking connect call and am checking for `NSAPI_STATUS_GLOBAL_UP` to check whether I'm connected. After a connect, I wait for a couple of seconds before disconnecting. Then again waiting a couple of seconds before starting with the connect again. See the code snippet below for what I'm doing as test, where `NetworkConnection` is a simple class around `CellularContext` / `NetworkInterface`. 

This is a simplified example from what I'm trying to do in my application

1. Do some measurements or whatever
2. Connect to the network
3. Send some stuff
4. Disconnect from the network
5. Go back to 1

See the attached log ([0113.log](https://github.com/ARMmbed/mbed-os/files/2880754/0113.log)) where I have tracing and AT commands enabled. Also see [network_connection.txt](https://github.com/ARMmbed/mbed-os/files/2880747/network_connection) for that class.

I don't understand why it would only do this section when I'm connecting for the second time.

```
2019-01-01 13:17:57 [16] Connecting... (submodules/mbed-system/source/main.cpp:334)

[INFO][CELL]: CellularContext set blocking 0

[INFO][CELL]: CellularContext connect

[DBG ][CELL]: checked operation 0 (0)

[DBG ][CELL]: returning
```

```c++
TRACE_DEBUG("Starting network connection test...");
    mbed_trace_init();

    wtg::connectivity::NetworkConnection network_connection;

    for (std::uint8_t i = 0; i < 50; i++) {
        TRACE_DEBUG("Connecting...");
        if (network_connection.Connect()) {
            TRACE_DEBUG("Connected to network: %s", network_connection.IpAddress().c_str());
        }
        TRACE_DEBUG("Done connecting! (%hhu)", network_connection.GetConnectionStatus());

        rtos::ThisThread::sleep_for(5000);

        TRACE_DEBUG("Disconnecting...");
        network_connection.Disconnect();
        TRACE_DEBUG("Done disconnecting! (%hhu)", network_connection.GetConnectionStatus());

        rtos::ThisThread::sleep_for(20000);
        PRINTF("\n\n\n\n\n");
    }

    TRACE_DEBUG("network connection test finished!");
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-02-19:
@ARMmbed/mbed-os-wan Please take a look.

vidavidorra, 2019-02-20:
An interesting thing here is that when I remove `_interface->set_blocking(false);` in the NetworkConnection (line 44). Every third network connection is able to make a connection. The latter is due to an incorrectly released semaphore (see [this](https://github.com/ARMmbed/mbed-os/issues/9412#issuecomment-460946908)).

vidavidorra, 2019-02-20:
What I also noticed today is that when I disconnect the antenna from the u-blox SARA-G350 module, it cannot connect while the antenna is disconnected. That is what I'd expect in this case, but when I plug the antenna back in, it is still not able to make a connection. After I reboot my device it immediately is able to create the connection again.
Could it be that this is due to the state that the state machine is in and it is not equipped for the antenna being removed? Seems odd to me if it is, but could be the case.

mirelachirica, 2019-02-20:
I have looked a bit on this issue and at leas one thing that seems to be problematic, but i need to check more, is that state machine is not reseted on disconnect, even if it should. 
I tried to add following:
`_device->cellular_callback(NSAPI_EVENT_CONNECTION_STATUS_CHANGE, NSAPI_STATUS_DISCONNECTED); `in  `AT_CellularContext::disconnect()` before `call_network_cb(NSAPI_STATUS_DISCONNECTED);`

Would you try if this is helping any of your cases?

mirelachirica, 2019-02-20:
void CellularDevice::cellular_callback() is ready to reset state machine on disconnect:

```
if (ev == NSAPI_EVENT_CONNECTION_STATUS_CHANGE && ptr == NSAPI_STATUS_DISCONNECTED) {
                // we have been disconnected, reset state machine ...
                if (_state_machine) {
                    _state_machine->reset();
                }
}
```

, but cellular context is not notifying the device on **disconnect**. That is what I suggest to do above.

vidavidorra, 2019-02-20:
@mirelachirica Thanks for your response.
To be clear, do I need to add the `_device->cellular_callback` before [this line](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularContext.cpp#L685) or all `call_network_cb(NSAPI_STATUS_DISCONNECTED);` calls in `AT_CellularContext.cpp`?

If you clarify, I'll tests and post back my results.

mirelachirica, 2019-02-20:
@vidavidorra I meant only in one place, in disconnect routine. Proper solution might require in all places but I propose to try in this one place first.

vidavidorra, 2019-02-20:
I didn't see a noticeable difference in behaviour, unfortunately. I do notice some extra logs (`AT OoB`)  after my first disconnect call. See the log attached.
[0124.log](https://github.com/ARMmbed/mbed-os/files/2885338/0124.log).
I've tried adding the `_device->cellular_callback` before all  `call_network_cb(NSAPI_STATUS_DISCONNECTED);` calls in `AT_CellularContext.cpp`, but again no noticeable difference.

mirelachirica, 2019-02-21:
The change I recommended earlier makes a difference only before [commit 63fb35](https://github.com/ARMmbed/mbed-os/commit/3bcf51385e50cc5f97b1c56f6495d9ce9397f535#diff-63fb35cdb5368ca46c1dd22806859b2f ) where the `call_network_cb(NSAPI_STATUS_DISCONNECTED);` was moved under `else` branch.

When this commit is in use can you try to take [call_network_cb(NSAPI_STATUS_DISCONNECTED);](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularContext.cpp#L685) out of else and add the `_device->cellular_callback` before it? Does this change have any impact?

mirelachirica, 2019-02-21:
@jarvte can you advice @vidavidorra on the following question:
"Also I really want to have the connection being set-up the same way it does the first time. I want to completely power down the modem and power it back on when I need to make a cellular connection again. How can I achieve this?"

vidavidorra, 2019-02-21:
> The change I recommended earlier makes a difference only before [commit 63fb35](https://github.com/ARMmbed/mbed-os/commit/3bcf51385e50cc5f97b1c56f6495d9ce9397f535#diff-63fb35cdb5368ca46c1dd22806859b2f) where the `call_network_cb(NSAPI_STATUS_DISCONNECTED);` was moved under `else` branch.

I am on version 5.11.1 so I'm definitely before that commit. Are there any significant differences in the cellular interface after version 5.11.1? I could update to 5.11.4 or 5.11.5-alpha if you think that helps with this issue.

> When this commit is in use can you try to take [call_network_cb(NSAPI_STATUS_DISCONNECTED);](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularContext.cpp#L685) out of else and add the `_device->cellular_callback` before it? Does this change have any impact?

Based on the above, I don't need to test this right?

vidavidorra, 2019-02-21:
@mirelachirica I see the following in the 5.11.2 release notes, could that be partially related to the issues I'm having?

> 9152
> Cellular: fix state machine to check network attach

mirelachirica, 2019-02-21:
> I am on version 5.11.1 so I'm definitely before that commit.

Oh, sorry I missed that. I was running my tests on recent master but before that commit. The change I suggested made the difference in that case but not for that commit. And since the link you indicated for the line `call_network_cb(NSAPI_STATUS_DISCONNECTED); `pointed to latest version and showed it under the else branch I thought that is the reason.

In 5.11.1 you have `call_network_cb(NSAPI_STATUS_DISCONNECTED);` outside of else branch already. There must be other reasons, then, for the problem.

I will check the other releases and see if I get any other clues about the problem.

vidavidorra, 2019-02-21:
@mirelachirica I've just updated to Mbed OS 5.11.4 to include 
> 9152
> Cellular: fix state machine to check network attach

and didn't note a difference in behaviour. See [0130.log](https://github.com/ARMmbed/mbed-os/files/2890015/0130.log).

<br>

> The change I recommended earlier makes a difference only before [commit 63fb35](https://github.com/ARMmbed/mbed-os/commit/3bcf51385e50cc5f97b1c56f6495d9ce9397f535#diff-63fb35cdb5368ca46c1dd22806859b2f) where the `call_network_cb(NSAPI_STATUS_DISCONNECTED);` was moved under `else` branch.

Based on that commit or other commits in master, could it possibly make a difference if I move from 5.11.4 to master? I saw in the commit you mentioned that some other things in the cellular also have changed. If you think it could make a difference, I can try to update to master to see if that makes difference.

vidavidorra, 2019-02-21:
@mirelachirica Also, I tried adding `GetDevice()->init_module();` on top of my connect function and am actually reading the module name (`SARA-G350`) back from the module. So, my GSM module is powered on properly only the cellular interface is somehow unable to go to the proper state. Really looks like an issue with the state machine or something.

vidavidorra, 2019-02-22:
@mirelachirica I think I might just have found something. I'm looking in more detail to the `CellularStateMachine` and all states it goes through and think I found a bug.
`CellularStateMachine::pre_event` gets the target state as `state` argument and checks if it needs to something to achieve the target state.
On [line 544](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/device/CellularStateMachine.cpp#L544) in `CellularStateMachine::pre_event`, shouldn't it check for `if (_state < state) {` rather than `if (_target_state < state) {`? Because it should compare check if it needs to do something based on the current state the machine is in and the desired (target) state, right? I'm no expert at how the cellular interfaces and state machine is intended to work, so I'm not 100% sure this is an actual bug but please check.

I've made that change in my code and found that it did change the behaviour a lot! It now actually tries to connect on the second connect call and goes through a lot more of the previous states. I'm now running into the issue that an error (`mbed_error`) is triggered somewhere on the second connect, so am now looking into why that is.
See my log attached [0134.log](https://github.com/ARMmbed/mbed-os/files/2893690/0134.log).

vidavidorra, 2019-02-22:
Okay, the error I'm getting is an stack overflow on the `stm_queue` thread.
```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: 
Location: 0x805576B
Error Value: 0x1
Current Thread: stm_queue  Id: 0x2000B138 Entry: 0x8058595 StackSize: 0x800 StackMem: 0x2000B190 SP: 0x2001FF2C 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020125
-- MbedOS Error Info --
```

vidavidorra, 2019-02-22:
I determined the required size of the `stm_queue` thread needs to be at least 2104 bytes. Currently it is 2048 bytes while the actual used bytes in my case is 1744 bytes. That is 304 bytes overhead in my case.
I tried to use the nearest 128 byte multiple for the size, since it used 2104 bytes, this is 2352 bytes (with an overhead of 248 bytes). I don't know if that is enough for all cases, but I'm currently running the code perfectly fine with 2352 bytes stack for the `stm_queue` thread.

mirelachirica, 2019-02-22:
@vidavidorra thanks for the great findings!

@jarvte could you check this please?

> On [line 544](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/device/CellularStateMachine.cpp#L544) in `CellularStateMachine::pre_event`, shouldn't it check for `if [...]

vidavidorra, 2019-02-25:
[0140.log](https://github.com/ARMmbed/mbed-os/files/2899494/0140.log)
[0141.log](https://github.com/ARMmbed/mbed-os/files/2899497/0141.log)

jarvte, 2019-03-01:
@vidavidorra can you test this pr: https://github.com/ARMmbed/mbed-os/pull/9898

vidavidorra, 2019-03-01:
@jarvte Will do, probably next Monday.

PR #9898 (2019-03-01): Cellular: fix connect-disconnect sequence called many times
### Description
Fix connect-disconnect sequence called many times

- Fixes #9760 
- Fix syncing back to at mode after ppp disconnect.
- Fix AT_CellularContext flags and states to allow new connect after disconnect.
- Fix that state machine is not reseted in disconnect is it's running (might be
  running because of another context or new connect already started).
- Removed API get_connection_status() from CellularNetwork. This was left accidentally
  after refactoring. It wasn't giving correct states after refactoring.

### Pull reques [... cut]
closed 2019-03-01 by 0xc0170