ARMmbed/mbed-os #9770: RawSerial with deepsleep not able to transfer

sivasanath, 2019-02-20:
### Description
Below program prints fine
RawSerial temppc(USBTX, USBRX,115200);
int main()
{

	unsigned int i=0;
	char str[100];

    while (true) {
        led1 = !led1;
		itoa(++i, str, 10); //Converts integer to string
		temppc.puts(str); 
		temppc.puts(" ");

		if(!(i%10)) {
			temppc.puts("\n");
		}
                sleep_manager_lock_deep_sleep();	
                wait(0.1f);
                sleep_manager_unlock_deep_sleep();
    }
}
the output is 
1 2 3 4 5 6 7 8 9 10
11 12 13 14 15 16 17 18 19 20
21 22 23 24 25 26 27 28 29 30
....
....
...
So on ...

But the below code prints junk characters, any idea why ?

RawSerial temppc(USBTX, USBRX,115200);
int main()
{

	unsigned int i=0;
	char str[100];

    while (true) {
        led1 = !led1;
		itoa(++i, str, 10); //Converts integer to string
		temppc.puts(str); 
		temppc.puts(" ");

		if(!(i%10)) {
			temppc.puts("\n");
		}
		wait(0.1f);
    }
}

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-02-20:
Which platform?

The problem will be that the platform's HAL is not waiting for any ongoing serial transmission to complete before entering deep sleep. The deep sleep is presumably stopping the serial UART's master clock, possibly mid-character, corrupting it.

There are no higher-level mechanisms in place to hold off on the clock shutdown, so it's currently up to the HAL to deal with this - basically for each active UART, wait until idle. There is currently no HAL API to obtain this information, so it would have to be done inside the HAL.

That wait could either be a block, or could be interrupt-driven - if `hal_deepsleep` sees there is ongoing serial TX, shallow sleep instead of deep, and enable an IRQ for "TX done". A done interrupt will wake from shallow sleep, and next sleep will be deep. But again, would have to be internal to the HAL.

The problem is that the HAL `serial_putc` is treated as a "fire and forget" synchronous blocking call, which generally returns the instant the data is put into the FIFO or holding register , so the main system has no idea whether serial data is still being clocked out. There's a HAL call for `serial_writable`, (it can accept more data), but not for `serial_tx_idle`.

I may be wrong, but I believe this necessary work is one of the main blockers to turning on `MBED_TICKLESS` on various platforms - serial corruption like this shows up immediately if the HAL doesn't have some mechanism to prevent it.

sivasanath, 2019-02-21:
Thanks @kjbracey-arm , got your point, we will be looking at a way on how to handle this in our serial hal driver.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@sivasanath Please reopen if you have further questions.