ARMmbed/mbed-os #9771: Cordio BLE stack: No callback comes to the application for Descriptor Read and Write

akkishore1, 2019-02-20:
When a GATT client reads or writes to the Characteristic descriptor application does not receive any read and write callback.
 
The below function GattServer::insert_descriptor in CordioGattserver.cpp does not set callback bits. Can you please check. Probably below bits need to be set for descriptors as well.

attribute_it->settings = ATTS_SET_READ_CBACK;
attribute_it->settings |= ATTS_SET_WRITE_CBACK;

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

desmond-blue, 2019-02-21:
Duplicate of #9732 
@akkishore1 We already merged the fix by this pull request #9748, could you check if this solve your issue? Thanks.

akkishore1, 2019-02-21:
Hi, This issue is different from #9732. #9732 is for Readcallback for Characteristics. 
This issue is there are no callbacks coming when we read or write to characteristic descriptors. I think you need to have similar callback bits set in GattServer::insert_descriptor  function in CordioGattserver.cpp

desmond-blue, 2019-02-21:
I apologize that I misunderstood this description, we will look into it.

desmond-blue, 2019-02-26:
Hi @pan- ,

My thought for what we need to change is

- Regarding Characteristic Extended Properties(read only), we need to add
`attribute_it->settings |= ATTS_SET_READ_CBACK;` if it's readable.

- Regarding Characteristic User Description and Server Characteristic Configuration, we need to add 
`attribute_it->settings |= ATTS_SET_READ_CBACK;` if it's readable.
`attribute_it->settings |= ATTS_SET_WRITE_CBACK;` if it's writable.

pan-, 2019-02-26:
@desmond-blue IIRC correctly it is asked that users set their properties for these descriptor types; the only exception being CCCD as they are implicitly instantiated if a characteristic is able to notify or indicate. More generally, I believe the whole GattServer API needs a refresh as it is not built on sane grounds.

desmond-blue, 2019-03-04:
@pan- 
Even if users set the read/write properties, current `GattServer::insert_descriptor()` API doesn't check this and doesn't set callback. I would like to propose [this fix](https://github.com/desmond-blue/mbed-os/commit/070a412432d736cfaca2b32153c11b366f10a2bb), would you take a look at it? 

Or do you already have a plan for refreshing this API?

Thanks for your help.

pan-, 2019-03-04:
@desmond-blue  Happy to approve your fix as it solves one of the issue. It must be stretched out that it does not solve the case when the value of the descriptor being read needs to be generated at access time or authorisation/rejection of a descriptor write. For that we would need an API refresh.

desmond-blue, 2019-03-06:
Thanks @pan- , so we could use this PR #9948 for this issue now, and we may need to raise other tasks for GattServer API refreshing?

akkishore1, 2019-04-05:
Fixed on master branch of mbed. Verified the same.

closed 2019-04-05 by akkishore1