ARMmbed/mbed-os #9773: Trigger the correct setTxPower method

antoniovlasic, 2019-02-20:
I've been searching for weeks now. I'm using a NRF52-DK (NRF52832) with the BLE Beacon example. 
When I call the Gap::setTxPower method it returns BLE_ERROR_NOT_IMPLEMENTED exception of course. Now I've seen there's a target implementation in mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_SOFTDEVICE/TARGET_NRF52/source.

But my problem is: I have absolutely no clue how to call the setTxPower method from the nRF5xGap class.

I'm sorry if the question is stupid, but weeks have passed since I've started to figure out what's wrong.

Best regards and thank you in advance

antoniovlasic, 2019-02-21:
Actually, my issue is: I don't understand how to target my NRF52832 in the code. I don't get what I'm doing wrong.

desmond-blue, 2019-02-22:
The setTxPower function was not standard until Bluetooth 5 and we expose a compliant API for it [here](https://github.com/ARMmbed/mbed-os/blob/f61dee1626b6e5c40905948545cafa0c9e87b73e/features/FEATURE_BLE/ble/gap/AdvertisingParameters.h#L459).

We have a [Mbed Question](https://os.mbed.com/questions/) area for raising usage questions, it's a better place for this kind of questions, and it's also good for other users with same questions to search.

Make sure you label the questions with "BLE", that helps us to characterize those questions.

Thanks,
Desmond

yogeshk19, 2019-03-05:
In the earlier versions of the BLE API, prior to BLE 5, you could Set the Tx Power irrespective if you are scanning or advertising. In the new API, the setTxPower seems to part of AdvertisingParameters class. In the old API, you could find out the valid Tx Power values and set one of those values. How do we achieve that with the new Cordio BLE API.

```
AdvertisingParameters &setTxPower(advertising_power_t txPower)
    {
        _txPower = txPower;
        return *this;
    }
```
Thanks,
Yogesh

desmond-blue, 2019-03-06:
Since the feature of setting Tx Power (scanning or advertising) is not defined in standard HCI of BLE spec, so it's not exposed to application level for users currently. We now are aware of this requirement and will schedule a task for analyzing this feature, it's not a standard feature so it must be done carefully.

Thanks for the comments.

Blackfire-Foo, 2019-07-01:
Where are we about this issue now, any chance get it fixed in the near future?
Or any workaround about "How can I increase (change) BLE TxPower if I use Legacy rather than Extended mode? I'm using nRF52480 + Cordio LL. Anyone can help with this?"
I keep searching answer for this question for several days now...
Thanks in advance.

0xc0170, 2019-07-02:
cc @ARMmbed/mbed-os-pan

yogeshk19, 2020-01-30:
@0xc0170, @pan-  Is there any progress on this bug? Is there a way to set tx power, when using the LEGACY_ADVERTISING_HANDLE? Plus I am not sure how the current API advertising_power_t which is an int8_t and allows values from -127 to 126. how does this map to the dbm values of nordic which has a maximum of 8 dbm for the NRF52840 BLE module. Also if you don't set the txPower, what does it default CORDIO platform set the Tx power of the BLE module to?

Thanks,
Yogesh

cblack-f3, 2020-01-30:
I am trying to figure this out too, it seems a bit complicated.  Since this is now part of ble::AdvertisingParameters, does that mean the power setting only applies to advertising?  What about TX power levels during connections?

paul-szczepanek-arm, 2020-02-20:
the spec only allows for control over advertising, anything else is vendor specific

yogeshk19, 2020-02-20:
@paul-szczepanek-arm I don't know why it would be under advertising? Most of the time the BLE devices in the field where sensors communicate with a say a central gateway would be in connected state than advertising state, when configured correctly by the configuration parameters in connected state, you get the lowest power draw on modules like NRF52840. Cordio Platform although has made programming a lot easier, is missing features such as simple as altering the transmit power and has heavy current draw issues. It's unfortunate now the Cordio Platform is not managed by the ARM/MBED team, and so issues such as this and current draw issue is just sitting there without any resolution in sight.

paul-szczepanek-arm, 2020-02-21:
> I don't know why it would be under advertising?

That's just how BLE spec is written.

You can do most things without a connection, maybe you can consider periodic advertising? The only time you need a connection is if both sides need to send data, otherwise being not connected will save more power. Establishing connections is fast.

0xc0170, 2020-02-21:
> the spec only allows for control over advertising, anything else is vendor specific

@paul-szczepanek-arm shall we close this issue as resolved?

paul-szczepanek-arm, 2020-02-21:
The original question was answered. Anything else please open a new issue and we'll try to answer. The values for the power settings are dbm
```
/**
 * Describe the advertising power.
 *
 * Values between -127 and +126 are considered power values in dBm while
 * the special value 127 can be used as a wildcard to indicate that the host
 * has no preference or if the power information is not available.
 */
typedef int8_t advertising_power_t;
```

closed 2020-02-21 by 0xc0170