ARMmbed/mbed-os #9788: cmsis.h missing include of cmsis_nvic.h for target TARGET_CM3DS_MPS2

itayzac, 2019-02-21:
### Description
File `./targets/TARGET_ARM_SSG/TARGET_CM3DS_MPS2/device/cmsis.h` does not include `cmsis_nvic.h` file. As a result, file `./rtos/TARGET_CORTEX/mbed_boot.c` does not contain a definition for `NVIC_RAM_VECTOR_ADDRESS` and function `mbed_cpy_nvic` performs nothing. 

The effect is that code `SCB->VTOR = (uint32_t)NVIC_RAM_VECTOR_ADDRESS` (in function `mbed_cpy_nvic`) will not be performed and vector table is not changed to target code, and remains at 0x0.

**tag: mbed-os-5.10.4**
**SHA: 2fd0c5c**
**toolchain: GCC_ARM**
**mbed-cli version: 1.2.0**
**target: TARGET_CM3DS_MPS2**

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-02-21:
@kapi90 Please review

kapi90, 2019-02-21:
Hi @0xc0170, yes this is a known issue, and there's a PR in the make with a couple of bug fixes on CM3DS, including this one as well.

cmonr, 2019-02-25:
@kapi90 Do you know the PR(s) that fixed this?

@itayzac Would you mind updating to a later version of Mbed OS and retesting?
(Also, Mbed CLI is a bit behind as well)

bulislaw, 2019-05-07:
@kapi90 is that fixed now?

adbridge, 2020-10-15:
As there has been no response in over a year to the questions asked here, I am closing this issue. If this is still a problem please raise a new one.

closed 2020-10-15 by adbridge