ARMmbed/mbed-os #9789: post_binary_hook in custom_targets.json

janjongboom, 2019-02-21:
### Description

post_binary_hook is looking in `mbed-os/tools/__init__.py` for functions to call. Is it possible to use this function in a custom_targets.json without having to modify this file? I want the binary hook to live in the same repository as my project and not modify Mbed OS itself.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

desowin, 2019-02-22:
Example problem is the disjoint flash sections in the user provided linker file. For example, if user provides his own linker file which includes the QSPI section on STM32F746 target, then the resulting bin file is over 2 gigabytes (MCU flash startings at 0x08000000 while the QSPI starts at 0x90000000). 

Hopefully, it is possible to change the output format to hex which produces much smaller file if the resulting elf has disjoint flash sections. Example `custom_targets.json` that changes the output format from bin to hex, leaving the rest of configuration intact is
```
{
    "CUSTOM_DISCO_F746NG": {
        "inherits": [
            "DISCO_F746NG"
        ],
        "OUTPUT_EXT": "hex"
    }
}
```

janjongboom, 2019-02-25:
@jeromecoutant @bulislaw How would we fix something like this?

bulislaw, 2019-02-27:
@bridadan something you could help with?

janjongboom, 2019-04-24:
@cmonr @bulislaw Why is this closed?

closed 2019-10-31 by ciarmcom