ARMmbed/mbed-os #9803: LoRa: Radio re-initialization possible?

maxgerhardt, 2019-02-21:
### Description

I'm having the following situation: Consider a LoRaWAN node. In my example, a Nucleo L476RG with an RFM95W (SX1276 style) radio. The node is powered and starts to send and receive LoRaWAN message. 

Now due to some condition, only the radio loses power and shortly after it regains power. (Or I switch off the power-supply for the radio using a load-switch for 0 power consumption.) 

What I have noticed with mbed-os's LoRaWAN stack and the Semtech drivers is that the firmware doesn't notice this short power outage. The next LoRaWAN message is queued and sent out. The `TX_DONE` event comes up, receive windows are opened et cetera. However, I have noticed that from that point on (power is shorty removed and restored to the radio), the uplink and downlink packet success rate is extremely worsened and never recovers. Only pressing the reset-button on the board lets it recover *immediately*; then nearly every sent packet is seen by the gateway again, where as a short period of time before, none were seen. 

In my test e.g., after sending 5 messages which are all seen by the gateway, after the power outage, most of the following packets (~90%) are not seen by the gateway. There is still an occasional packet coming through, but the ratio is very low. Something seems just broken after that.

My requests / questions:
* can you reproduce that? (e.g. take https://github.com/ARMmbed/mbed-os-example-lorawan as the firmware and an [RFM95W radio breakout](https://www.adafruit.com/product/3072) with a Nucleo L476RG) If yes, this implies the driver / MAC doesn't survive a power-glitch and the node is essentially dead from that point after if no measures are taken in usercode.
* why is initialization of the radio only in the constructor and never during the runtime or exposed as a function? 

The only time `radio_init()` is called is in the constructor of the `LoRaWANInterface` object, this subsequently calls into [`init_radio(radio_events)`](https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers/blob/master/SX1276/SX1276_LoRaRadio.cpp#L243) with an object held by the LoRaWAN interface. Thus from the outside as a user, you can't re-initialize the radio yourself because you don't have `radio_events`. Even calling into `lorawan.disconnect()` and `lorawan.connect()` will not trigger a radio-reset and re-initialization. Can't this be exposed to the user in order to e.g. react to a `TX_FAILURE` and friend events from the LoRaWAN stack, or as a per-se safety measure?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

maxgerhardt, 2019-02-24:
Appendix: The problem seems, most interestingly, to only occur on higher SF (I use SF12, no ADR). The problem can be fixed by deleting the `LoRaWANInterface` object and re-creating it. The next transmission after this reset will go through again as normally.

```cpp
extern SX1276_LoRaRadio radio;

void LoRaWAN_Reset() {
    delete lorawan;
    lorawan = new LoRaWANInterface(radio);
    //function which calls into lorawan->initialize() and conenct() etc
    LoRaWAN_Init();
}
```

maxgerhardt, 2019-03-04:
This is still an issue. The workaround above has the problem that with the `lorawan` object, all info about duty cycle back-off times etc are destroyed, so it's effectively switching off duty cycling. The mbed-os driver should work when the radio's power is switched off by user, that's what I'd expect. Any updates on the reproducibility from your side? 11 days of radio silence :( @cmonr?

maxgerhardt, 2019-03-05:
Also somehow the framecounters are destroyed so they always send `fCnt = 0` which destroys the purpose of frame counter validation.

hasnainvirk, 2019-03-11:
@maxgerhardt It's an interesting but odd case. Ofcourse the radio driver was not designed to cope with  external power cuts like you mentioned. 
The lowest power state for SX1276 is sleep (there is no shutdown/powerdown state). When the radio is in sleep state PLL, radio circuitry, amp and other chain components are off. Datasheet says it should consume about 0.1 to 1 uA. 
Now you are physically turning off the chip that means the chip will not retain any register state. All configurations are lost. 

Deleting LoRaWANInterface alone will not achieve much for you. You got to delete LoRaRadio object too. And yes doing that will cause grief as frame counters and any state including DC  will be lost. So is the case when you hit a reset button. Majority of the people will be happy with 0.1 uA probably in sleep state, however in corner cases like yours, I am foreseeing a tad bit over engineering. Nonetheless, it is very interesting and worth further investigation. I was wondering if you have/could measure, how much current the chip draws when you ramp the power up after physical disconnection of power ?

maxgerhardt, 2019-03-14:
@hasnainvirk The more important part to me is the 'safety' aspect, 'lowpower' is second here. The tricky problem is that if a momentary (partial) power-loss occurs, the driver will happilly send out messages after that, but in my tests these messages (up & downlink) barely get through; so the connection is somehow crippled, but undetectably so (if e.g. the application is only supposed to send uplink data). And this must be due to the radio reverting some register settings back to reset-on-power values, I presume. However this would implies that the TX/RX routines don't fully setup the radio. 

That's why I'd be really keen on seing if you can reproduce the issue as I have written above using the simple mbed-os-example-lorawan application on SF12.

> You got to delete LoRaRadio object too 

Why? The destructor does nothing (except kill the thread when the object is deletecd) and the constructor only does pin assignments and start the radio thread, so I don't see anything important in there ([code](https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers/blob/master/SX1276/SX1276_LoRaRadio.cpp#L159-L211))

Actually the 'fix' I'm searching for this thus only a reset routine which will pulse `RST` and re-setup registers so that work may continue. However I haven't been able to produce such a fix yet.

0xc0170, 2019-08-14:
> Actually the 'fix' I'm searching for this thus only a reset routine which will pulse RST and re-setup registers so that work may continue. However I haven't been able to produce such a fix yet.

@maxgerhardt Have you made any fix? Is this issue still valid as it has not been updated for months.

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by ciarmcom