ARMmbed/mbed-os #9809: QSPIBlockDevice memory mapped mode support

desowin, 2019-02-22:
### Description

QSPI peripherals in multiple supported targets (eg. STM32F7, STM32F4) support emmory mapped mode. Unfortunately the QSPIBlockDevice does not support switching to memory mapped mode (which atleast  in case of STM32 is read-only). Example use of memory mapped mode is the graphics storage for devices with LCD display. The memory mapped modeis in fact the recommended mode when using TouchGFX library. I would like to request a function in QSPIBlockDevice that would switch the device to memory mapped mode. The QSPIBlockDevice could be then aware of the memory state and eg. fail all write requests if the QSPI is memory mapped in STM32 targets.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

desowin, 2019-04-18:
Would it be possible to know the reason why this was closed?

0xc0170, 2019-04-18:
The issue should always have a reason for closure, sorry about that.

@linlingao Please provide it

linlingao, 2019-04-18:
This was accidentally closed by a script.

0xc0170, 2019-08-14:
I believe this PR https://github.com/ARMmbed/mbed-os/pull/11006 will close this request

desowin, 2019-08-15:
> I believe this PR #11006 will close this request

Although the PR is strongly related, it is not quite the same. The use case mentioned in the issue description does not require XIP functionality at all. The use case is only to enable the data to be accessed directly by the Chrome-ART 2D DMA engine. I'd also like to have the mapMemory() and unmapMemory() functions. the map/unmap is necessary if the application wants to write to the memory (eg. half of the QSPI flash is graphics storage while the other half is used for logs; QSPI flash write is not supported in the memory mapped mode). In such case application needs to perform appropriate locking (so the graphics task does not try to access the memory when it is unmapped).

closed 2019-10-31 by ciarmcom