ARMmbed/mbed-os #9816: Export to GCC missing mbed_config.h

hierophect, 2019-02-22:
Make-GCC-ARM exports of all MBED os programs fail to compile with Make, including the Blinky-OS example, citing the following error:
Compile: main.cpp
Assemble: except.S
cc1: fatal error: /filer/workspace_data/exports/8/8337c13910a897ed961244fa8538db7e/mbed-os-example-blinky/mbed_config.h: No such file or directory
compilation terminated.
make[1]: *** [mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_GCC/except.o] Error 1
make: *** [all] Error 2

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

linlingao, 2019-02-22:
Please specify exactly which blinky example. There're multiple blinky examples online. Please also specify the target you're building.

hierophect, 2019-02-22:
I'm referring to the default blinky example provided by the MBED online compiler, mbed-os-example-blinky, the default template suggested for any mbed-os program. I've had this error for all of my development boards, which include the NUCLEO-F401RE, the NUCLEO-F303K8, the NUCLEO-F411RE, and the NUCLEO-F103RB.

linlingao, 2019-02-25:
I'm not able to recreate the issue.
My steps:
(1) Add NUCLEO-F303K8 to the online compiler.
(2) Inside Mbed online compiler, click New->New Program. Platform is prefilled with NUCLEO-F303K8, the default Template is Blinky LED test for the ST Nucleo boards. Program Name: Nucleo_blink_led. Click OK.
(3) Compile, a binary is generated.
(4) Highlight Nucleo_blink_led, Export Target is pre-filled with NUCLEO-F303K8, Export Toolchain set to MAKE-GCC-ARM. Click Export. Zip is generated.
(5) Unzip and import it into Eclipse IDE. Verify mbed_config.h in Project Explorer. Build the project.
I don't see any build error.
Please retry and let me know if you still experience issues.

hierophect, 2019-02-25:
Excuse me, I'm sorry. I mistakenly included the STM32F303K8, that was a typo on my part since I use it so frequently alongside my other boards - the 303K8 can't run mbed-os, only mbed-2, so the example you have tried is not actually an MBED-OS program. I have no problem exporting mbed-2 programs, just the mbed-os ones. 

To make sure, I've gone back and verified for the three boards I have that can run mbed-os, and I can confirm it still fails to compile for them, using the gcc-arm-eabi-none compiler on my mac machines. Exports of any of my mbed 2 programs work just fine. I'll try and verify whether the same issue occurs on my windows machine if I can get the arm-gcc compiler up and running.

linlingao, 2019-02-25:
I can recreate the issue now with NUCLEO-F401RE.
I'll get our team to fix the exporter. In the meantime, you can workaround this by making a simple change in the Makefile:
Line 1493 (or search for "filer")
Change the following line:
ASM_FLAGS += ../mbed_config.h

Thanks for reporting this issue.

hierophect, 2019-02-25:
Great, thanks!

bridadan, 2019-04-26:
This should be fixed after Mbed OS 5.12.2 goes out.

microbit-carlos, 2019-05-13:
I've exported a project from the online compiler today, it is using Mbed OS 5.12.3 and I still encountered this issue. Has everything needed for this fix been deployed?

linlingao, 2019-05-13:
@theotherjimmy Do you happen to know which PR was supposed to fix this?

linlingao, 2019-06-11:
https://github.com/ARMmbed/os.mbed.com/pull/1743 should address this, it's not merged yet.

rkreifeldt, 2019-07-13:
Any idea when this is going to be be fixed?

nat45928, 2019-07-26:
Also seeing this on the NUCLEO_F767ZI target.

0xc0170, 2019-08-14:
> This should be fixed after Mbed OS 5.12.2 goes out.

@ARMmbed/mbed-os-tools When we expect this to be fixed ? Referenced PR :

> ARMmbed/os.mbed.com#1743 should address this, it's not merged yet.

was closed?

mark-edgeworth, 2019-08-14:
I am sorry but we do not currently have the resource here to investigate this further. We'd welcome a PR with a fix for this should you be willing to contribute.

hierophect, 2019-08-14:
Mbed doesn't have the resources to fix a simple but universal bug with their primary export tool? It's literally a line in the makefile. Are you guys shutting down? 

I'd be more inclined to do the footwork for a PR if that message wasn't so ominous...

madchutney, 2019-08-14:
Apologies for the inconvenience and understand your frustration. I can assure you that Arm and Mbed are not going anywhere and we will continue to develop Mbed, support the community and ecosystem. However, this is an open source project and we do not have resources to do everything, which is why we encourage contributions.

hierophect, 2019-08-14:
@madchutney I understand the importance of PR contribution, but given that this is a compatibility bug with the exporter in the online compiler, it's not clear to me how a community member would be capable of testing a solution to this issue. Any PR efforts would be shooting in the dark (if the export scripts are exposed at all, which I'm not aware of). Conversely, for an internal team member, shouldn't this be as simple as adding the `ASM_FLAGS += ../mbed_config.h` to the makefile generation in the exporter? That seems like a very quick fix.

loverdeg-ep, 2019-08-14:
@hierophect Lets just hope Mbed Studio gets web-hostable sooner than expected.

Hopefully this means goodbye to Mbed OS 2 as well.

rkreifeldt, 2019-08-15:
Looking at how terrible Mbed Studio is, I wonder why ARM Mbed is wasting their valuable resources on this. Why don't they just make a plugin for Eclipse? Then we'd have all the mature technology of Eclipse for experienced developers and the web version of MBED for those just getting their toes wet.

linlingao, 2019-08-15:
@rkreifeldt I'm sure our Mbed Studio team would like to hear the exact issues you have with Mbed Studio so that they can make improvements. Please feel free to send your feedback to studio@mbed.com.

madchutney, 2019-08-22:
@0xc0170 https://github.com/ARMmbed/os.mbed.com/pull/1743 was closed but the tools were updated to 5.13 here https://github.com/ARMmbed/os.mbed.com/pull/1938 and since this issue persists the fix can't have worked.

madchutney, 2019-08-22:
Hi @hierophect, firstly my apologies I hadn't realised this was an online issue only. I agree it is very difficult to work on an online compiler issue. The exporters are actually the same offline and online but this issue only manifests itself when invoked online, so tricky to test for me and impossible for you. I wish the fix was as simple as you suggest, however this would break the offline export from the mbed-cli.

I'll reopen the issue and hopefully get some time to fix it, again apologies for the misunderstanding.

lefebvresam, 2020-05-14:
I use VSC and Cygwin. I had to change line 1663 after exporting to vscode_gcc_arm from:
ASM_FLAGS += .\mbed_config.h
into
ASM_FLAGS += mbed_config.h
Bug is still there.

Abeldbest, 2020-12-05:
I have faced the same issue when exporting to Eclipse from the Online Compiler.
Since this issue has no perspective to be solved in the short term, why not update mbed documentation and let developers know this limitation.
I have wasted many hours finding this wire, just to modify a line of the Makefile.

loverdeg, 2020-12-10:
@Abeldbest there is too much legacy technical debt to pay. This project (Mbed) has been trying to evade said debt for years. There finally looks to be light at the end of the tunnel though. But not for this issue. You might as well consider the legacy online compiler dead.

0xc0170, 2020-12-10:
I'll close this as "won't fix".

> Since this issue has no perspective to be solved in the short term, why not update mbed documentation and let developers know this limitation.

As you already looked at the this, where did you look at first (which docs page) ?

closed 2020-12-10 by 0xc0170

JerryZheng89, 2021-01-13:
Today 2021, the problem still there!
when I export to vscode, the 1804 line of Makefile still wrong!
when you see this, change it to 
```c
ASM_FLAGS += ../mbed_config.h
```

lefebvresam, 2021-01-15:
One tip for Windows users, install WSL2 and use Visual Studio Code with WSL integration. It is 100% Linux on a Windows PC and no longer Windows/Linux conflicts in scripts.

bcsdriver, 2021-03-17:
I noticed that the problem is not the missing file but the wrong path generated by the export:

`ASM_FLAGS += /filer/workspace_data/exports/......../YOUR_PROJECT/mbed_config.h`

This shall be deleted and replaced by:

`ASM_FLAGS += ../mbed_config.h`

and then it compiles without this error.

lefebvresam, 2021-03-18:
> One tip for Windows users, install WSL2 and use Visual Studio Code with WSL integration. It is 100% Linux on a Windows PC and no longer Windows/Linux conflicts in scripts.

You can also use a Linux laptop somewhere on your desk where you develop and run commands trough ssh (from a wsl console window) and connect a remote Visual Studio Code session to enable programming from a Windows PC. This makes things much easier and you do not put addional cpu load your user laptop for compiling. In the meantime you can stay using all your Windows applications.