ARMmbed/mbed-os #9826: BLE NRF52840: RawSerial, Losing characters while reading from UART.

yogeshk19, 2019-02-24:
### Description
    - Target : NRF52840
    - Toolchain : GCC_ARM
    - mbed-cli
    - Mbed OS 5.11.2
   
We currently have a custom board that houses a NRF52840 BLE module and the application is built on top of Cordio BLE. We use RawSerial read/write via Max Chip on the custom board to read/write from an external device/sensor (RS-232 interface). When communicating with the device i.e. we write a command and the device responds back with data which is all plain ASCII string. However we are losing almost 90% of the ASCII data being returned.  So on a different custom board, that uses Multitech's MDOT we have a similar interface and it works without any issues, its just that we seem to run into issues when using this BLE module NRF52840. I am not sure where and how to begin troubleshooting this issue. Is there any additional settings required on the NRF52840?

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

Thanks,
Yogesh

yogeshk19, 2019-03-03:
It seems like the custom board doesn't seem to be able to operate under any baud rate other than 9600, so we are looking into it if this an issue with the custom board or there is some issue with the API, where we are unable to change the baud rate. I see a method available to set the baud rate in RawSerial class. However is there one to verify what baud rate was applied?

Thanks,
Yogesh

desmond-blue, 2019-03-04:
Hi Yogesh,

I add some serial code snippets into BLE example, and try read/write with pc, it works well over baud rate 9600/38400/115200.

[This ](https://github.com/ARMmbed/mbed-os/blob/59549b8eb40ebdf51b1f2454ac04156e149b8c54/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L688)is where the HAL sets baud rate, you may check the configuration here since we currently don't have an API for getting baud rate from the top level.

Which UART object are you using? The print function is using UART0 by default maybe there is a conflict in your configuration?

yogeshk19, 2019-04-19:
It was the external crystal that needed to be updated on the custom board. We are able to operate at higher baud rates now. I am closing this as resolved. Thanks for all your inputs.

closed 2019-04-19 by yogeshk19