ARMmbed/mbed-os #9830: PSOC6 targets not building for ARMC6

SeppoTakalo, 2019-02-25:
### Description

Cypress PSOC6 based targets fail to build when using ARMC6 compiler and `feature-armc6` branch.

```
09:46:10 [ERROR] In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CMOD_062_4343W/device/system_psoc6_cm4.c:28:
09:46:10 In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/devices/include/device.h:22:
09:46:10 In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/objects.h:29:
09:46:10 In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/gpio_object.h:27:
09:46:10 In file included from ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_gpio.h:177:
09:46:10 ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_syslib.h:564:12: warning: unknown attribute 'zero_init' ignored [-Wunknown-attributes]
09:46:10     extern CY_NOINIT cy_stc_fault_frame_t cy_faultFrame;    /**< Fault frame structure */
09:46:10            ^
09:46:10 ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/include/cy_syslib.h:480:69: note: expanded from macro 'CY_NOINIT'
09:46:10     #define CY_NOINIT           __attribute__ ((section(".noinit"), zero_init))
09:46:10                                                                     ^
09:46:10 ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CMOD_062_4343W/device/system_psoc6_cm4.c:526:7: error: expected '(' after 'asm'
09:46:10 __asm void Cy_MemorySymbols(void)
09:46:10       ^
09:46:10 ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CMOD_062_4343W/device/system_psoc6_cm4.c:526:6: error: expected ';' after top-level asm block
09:46:10 __asm void Cy_MemorySymbols(void)
09:46:10      ^
09:46:10      ;
09:46:10 
./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CMOD_062_4343W/device/system_psoc6_cm4.c:529:5: error: use of undeclared identifier 'EXPORT'
09:46:10     EXPORT __cy_memory_0_start
09:46:10     ^
09:46:10 1 warning and 3 errors generated.
09:46:10 
```

@ARMmbed/team-cypress  Please review.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

maclobdell, 2019-04-11:
@SeppoTakalo - can you confirm that this issue is resolved for you now?

vmedcy, 2019-04-11:
@maclobdell: #10308 is not yet merged to master, rework by Cypress team is in progress.

vmedcy, 2019-04-18:
This is fixed in the latest master, all PSoC 6 targets updated to ARMC6.

closed 2019-05-07 by ciarmcom