ARMmbed/mbed-os #9848: CELLULAR can't compile is DEVICE_SERIAL is not enabled

jeromecoutant, 2019-02-26:
### Description

How to reproduce: remove SERIAL from DEVICE_HAS in targets json file, and compile a target (with no cellular):

[Error] CellularDevice.h@197,0:  #20: identifier "UARTSerial" is undefined                                                                                                                                                                    [Error] CellularContext.h@269,0:  #20: identifier "UARTSerial" is undefined                                                                                                                                                                   [Error] AT_CellularContext.h@61,0:  #20: identifier "UARTSerial" is undefined                                                                                                                                                                 [Error] AT_CellularContext.cpp@92,0:  #20: identifier "UARTSerial" is undefined                                                                                                                                                               [Error] AT_CellularContext.cpp@105,0:  #20: identifier "UARTSerial" is undefined

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-02-26:
Even with .mbedignore fiel with:

````
*cellular*
*greentea*
*utest*
*unity*
````

We got link error:
````
[DEBUG] Errors: Error: L6218E: Undefined symbol CellularInterface::get_target_default_instance() (referred from BUILD/NUCLEO_L476RG/ARM/features/netsocket/NetworkInterfaceDefaults.o)
````

@SeppoTakalo

SeppoTakalo, 2019-02-26:
@AriParkkila This needs to go into Cellular teams backlog.

Those files needs probably `#if DEVICE_SERIAL` to protect the build.

jeromecoutant, 2019-02-26:
https://github.com/ARMmbed/mbed-os/blob/b08ddaad8b9db97eba3fea4e6b0caf6eb36943be/drivers/UARTSerial.h#L23

Seems that we need to protect also with DEVICE_INTERRUPTIN

Thx

jeromecoutant, 2019-02-28:
Hi

Note that this issue is raised during TFM scope. (#9221) (#9853)

cc @ARMmbed/mbed-os-hal @ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-psa

0xc0170, 2019-02-28:
@ARMmbed/mbed-os-wan

jeromecoutant, 2019-03-06:
Bump

Thx

AriParkkila, 2019-03-11:
@jeromecoutant Can you refer to an Mbed target that does not support `DEVICE_SERIAL` so I could try with that?

It seems to me that Mbed OS actually requires `DEVICE_SERIAL`, for example Greentea:
```
Compile [ 15.7%]: greentea_serial.cpp
[ERROR] In file included from ./features/frameworks/greentea-client/source/greentea_serial.cpp:1:
./features/frameworks/greentea-client/greentea-client/greentea_serial.h:10:37: error: expected class name
class GreenteaSerial : public mbed::RawSerial {
                                    ^
./features/frameworks/greentea-client/source/greentea_serial.cpp:15:42: error: member initializer 'RawSerial' does not name a non-static data member or base class
```

jeromecoutant, 2019-03-11:
Hi

> Can you refer to an Mbed target that does not support DEVICE_SERIAL so I could try with that?

This question is related to TFM integration.
Idea is to create a new secure target which doesn't need SERIAL API.

> It seems to me that Mbed OS actually requires DEVICE_SERIAL, for example Greentea

Yes, I raised #9850 :-)

AriParkkila, 2019-03-11:
@jeromecoutant #10029 fixes cellular dependencies to `DEVICE_SERIAL`.

Could you create an issue for `UARTSerial.h / DEVICE_INTERRUPTIN` to be fixed separately.

kjbracey-arm, 2019-03-11:
I believe Jerome's point in the comment above was that your compile condition needs to match `UARTSerial`'s availability condition - ie it has to be `DEVICE_SERIAL && DEVICE_INTERRUPTIN`.

Although that condition is itself rather strange - it's only the DCD line that it needs `InterruptIn` in for, so maybe it would be better if `UARTSerial` only made that DCD functionality conditional.

AriParkkila, 2019-03-11:
`DEVICE_INTERRUPTIN` might be needed in any case? For example K64F target has compile errors without `DEVICE_INTERRUPTIN`:
```
[Error] NanostackRfPhyMcr20a.h@64,11: 'InterruptIn' in namespace 'mbed' does not name a type

```

AriParkkila, 2019-03-19:
@jeromecoutant can we close this issue? PR #10029 was merged.

closed 2019-03-19 by jeromecoutant