ARMmbed/mbed-os #9850: GREENTEA can't compile is DEVICE_SERIAL is not enabled

jeromecoutant, 2019-02-26:
### Description

How to reproduce:
- remove SERIAL from DEVICE_HAS in targets json file
- add "\*cellular\*" in .mbedignore

````
$ mbed compile -t ARM -m NUCLEO_L476RG -v

[Error] greentea_serial.h@10,0:  #1046: namespace "mbed"  has no member class "RawSerial"
[Error] greentea_serial.h@10,0:  #262: not a class or struct name
[Error] greentea_serial.cpp@15,0:  #1046: namespace "mbed"  has no member class "RawSerial"
[Error] greentea_serial.cpp@15,0:  #18: expected a ")"
````

I don't see any reason why greentea-client should be compiled when DEVICE_SERIAL is not enabled.

Thx

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-02-26:
Note there is also impacts in:

- UTEST compilation:
````
[Error] utest_serial.h@25,0:  #5: cannot open source input file "greentea-client/greentea_serial.h": No such file or directory
````

- unity compilation: 
````
[Error] unity.c@8,0:  #5: cannot open source input file "utest/unity_handler.h": No such file or directory
````

jeromecoutant, 2019-02-28:
Hi

Note that this issue is raised during TFM scope. (#9221) (#9853)

cc @ARMmbed/mbed-os-hal @ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-psa

kjbracey-arm, 2019-03-11:
Somewhat related: #6674

PR #11086 (2019-07-25): Mbed-os compile without DEVICE_SERIAL
### Description

Added preprocessing `#if` statements so that Mbed-os can compile without DEVICE_SERIAL.
fixes #9850

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@jamesbeyond @maciejbocianski
closed 2019-07-25 by SeppoTakalo