ARMmbed/mbed-os #9853: USTICKER can not be removed from compilation

jeromecoutant, 2019-02-26:
### Description

Discussion started here:
https://github.com/ARMmbed/mbed-os/pull/9221#discussion_r258532501

@ARMmbed/mbed-os-psa 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-02-27:
We shall fix this asap as its blocking PR #9221

cc @ARMmbed/mbed-os-hal @kjbracey-arm

0xc0170, 2019-02-27:
Our Mbed Enabled defines ticker as a must: https://www.mbed.com/en/about-mbed/mbed-enabled/requirements/

ticker is considered as a basic functionality. If we have a valid use case, we might need to send a fix.

mikisch81, 2019-02-27:
From the Musca PR:

When moving `us_ticker.c` into the NS target I get the following linker errors:
```sh
Error: L6218E: Undefined symbol us_ticker_clear_interrupt (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_disable_interrupt (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_fire_interrupt (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_free (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_init (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_read (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Error: L6218E: Undefined symbol us_ticker_set_interrupt (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/hal/mbed_us_ticker_api.o).
Finished: 0 information, 2 warning and 7 error messages.
```

When looking into `mbed_us_ticker_api.c` I see that there is a static struct of function pointers - [us_interface](https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_us_ticker_api.c#L23:L32) which the linker will not throw, therefor even that the secure target does not use us_ticker at all, it needs to resolve those symbols - hence the empty implementation.
I agree this is not the right behaviour, but eventhough the target is using the bare-metal prfofile to build, `hal` folder is still built entirely.
@alzix @donatieng @theotherjimmy @bridadan @jeromecoutant

mikisch81, 2019-02-27:
It appears the problem is that [mbed_die()](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_board.c#L26-L49) implementation calls [wait_us()](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_wait_api_no_rtos.c#L35-L40) which calls `get_us_ticker_data()` which is defined in `mbed_us_ticker_api.c`.
So just guarding `mbed_us_ticker_api.c` with `#if DEVICE_USTICKER` is not enough.

mikisch81, 2019-02-27:
@orenc17 @alzix

kjbracey-arm, 2019-02-27:
The fact that `DEVICE_USTICKER` exists suggests you might be allowed to not provide it, but as far as I'm aware it is assumed that it must exist in various places, and we've not coded to code with a platform without it.  `DEVICE_LPTICKER` is certainly optional.

As a quick workaround you could conditionalise `wait_us` to have a fallback to the lp ticker if `DEVICE_USTICKER` is not defined, I guess.

mikisch81, 2019-02-27:
And if lp ticker is not defined as well?

kjbracey-arm, 2019-02-27:
So this is the secure side or something - super limited environment, so no timers at all?

In that case you'll could have a  condition in mbed_die to avoid the blinky LED that uses the timing. Or maybe we could have `wait_us` fallback to `wait_ns`, which is in a pending PR now and uses a software loop, calculating iterations based on SystemCoreClock.

0xc0170, 2019-02-27:
> In that case you'll could have a condition in mbed_die to avoid the blinky LED that uses the timing. Or maybe we could have `wait_us` fallback to `wait_ns`, which is in a pending PR now and uses a software loop, calculating iterations based on SystemCoreClock.

Was considering this, the units used now in mbed_die would be OK for `wait_ns` ?

mikisch81, 2019-02-27:
When is #9812 expected to be merged? 5.12.0 as well?

kjbracey-arm, 2019-02-27:
Conceivably you could unconditionally use `wait_ns` I guess, to avoid pulling in the `us_ticker` code, even if available. The timing isn't at all critical.

`wait_ns` can overflow its maths though - it's documented as "don't try passing more than 1000000". And we can't realistically make it use wider maths or the overhead becomes infeasible for something that's supposed to be a "nanoseconds" delay. The test case for `wait_ns` breaks it up by calling `wait_ns(1000000)` once for each millisecond.

mikisch81, 2019-02-27:
@kjbracey-arm I saw that `mbed_die()` uses `wait_us(400000)`, which means it should be `wait_ns(400000000)`.
But you said that we should pass bigger than 1000000, so what can we do then?

kjbracey-arm, 2019-02-28:
Call `wait_ns(1000000)` 400 times.

mikisch81, 2019-02-28:
@kjbracey-arm Maybe change the `wait_us()` implementation?
```c
void wait_us(int us)
{
#if DEVICE_USTICKER
    const ticker_data_t *const ticker = get_us_ticker_data();
    uint32_t start = ticker_read(ticker);
    while ((ticker_read(ticker) - start) < (uint32_t)us);
#else
    uint32_t iterations = us / 1000;
    uint32_t last_ns = us % 1000;
    for (uint32_t i = 0; i < iterations; ++i) {
        wait_ns(1000000);
    }
    if (last_ns > 0) {
        wait_ns(last_ns);
    }
#endif
}
```

kjbracey-arm, 2019-02-28:
Yes, that would make sense. I would save the division by just doing

    while (us > 1000) {
        us -= 1000;
        wait_ns(1000000);
    }
    if (us > 0) {
        wait_ns(us * 1000);
    }

Still, accuracy of `wait_us` will be greatly diminished like that. But I guess you don't expect timing accuracy on a platform with no timers?

mikisch81, 2019-02-28:
> 
> Still, accuracy of `wait_us` will be greatly diminished like that. But I guess you don't expect timing accuracy > on a platform with no timers?

Exactly.. therefor the ifdef #DEVICE_USTICKER

kjbracey-arm, 2019-02-28:
My concern is what is the caller of `wait_us` expecting? Do they *know* they're on a platform with no `DEVICE_USTICKER`?   We're kind of breaking a contract here for the application API.     I'm still not 100% sure what the use case here.

mikisch81, 2019-02-28:
The use-case are constrained devices (in our case the secure part of a PSA target).

I am completely fine with putting this logic in `mbed_die()` implementation.

kjbracey-arm, 2019-02-28:
Mind you, we do have existing examples of tolerating poor timing. There's a config option for STM targets that makes the lpticker be based on some circuit that has maybe 20% error rather than a proper 32.768kHz crystal...

kjbracey-arm, 2019-02-28:
I'm fine putting this in `wait_us` as long as we aren't calling any target with `DEVICE_USTICKER` off a suitable target for general-purpose user applications. If it's only happening for the secure side of PSA, that's fine.

closed 2019-03-14 by 0xc0170

mikisch81, 2019-03-18:
I still get errors building a secure target without USTICKER, now `Timer.cpp` is failing):
```sh
Error: L6218E: Undefined symbol get_us_ticker_data (referred from BUILD/ARM_MUSCA_A1_S/ARMC6/mbed-os/drivers/Timer.o).
```

kjbracey-arm, 2019-03-18:
I suggest you make the definition of `Timer::Timer()` conditional on `DEVICE_USTICKER`.

The ARM linker forces programs to obey the C++ "One Definition Rule", so that default constructor referring to `get_us_ticker_data` forces it to have one definition, even if the default constructor is never called. If the definition doesn't exist, it's not a legal C++ program:

> Every program shall contain exactly one definition of every non-inline function or variable that is odr-used in that program; no diagnostic required.

An alternative workaround would be to eliminate the default constructor and instead declare the other constructor with `Timer::Timer(const ticker_data_t *data = get_us_ticker_data())`, but messing with the constructor overloads might introduce binary compatibility issues we could do without.  That would work because naming something in an unused default parameter doesn't count as odr-using it, unlike having it in an unused function.

PR #9896 (2019-03-14): wait_us without usticker
### Description

Some targets (Mainly PSA secure targets running TF-M) have no usticker.
In order to support `wait_us()` in these targets (`mbed_die()` uses it for example), `wait_us()` is implemented using `wait_ns()` which does software-loop based delays.

* only last commit is relevant, other commits are from #9812.

Fixes #9853 

Depends on #9812 

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@kjbracey-arm 

### Release Notes