ARMmbed/mbed-os #9854: Compilation is failing is MPU is not enabled

jeromecoutant, 2019-02-26:
### Description

Ex with NUCLEO_L476RG

- remove MPU from DEVICE_HAS

````
$ mbed compile -t ARM -m NUCLEO_L476RG -v

[DEBUG] Output: ".\hal\mpu\mbed_mpu_v7m.c", line 25: Error:  #35: #error directive: "Device has v7m MPU but it is not enabled. Add 'MPU' to device_has in targets.json"
````

__MPU_PRESENT is defined in CSMIS file,
why do we need to redefine it in DEVICE_HAS json configuration ?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-02-28:
Hi

Note that this issue is raised during TFM scope. (#9221) (#9853)

cc @ARMmbed/mbed-os-hal @ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-psa

0xc0170, 2019-03-08:
@c1728p9 Wasn't this intention?

c1728p9, 2019-03-08:
Hi @jeromecoutant this error is intentional for two reason.
1. Porters adding new targets need to be aware that there is a common MPU API driver. Enabling this by default based on `__MPU_PRESENT` may lead to problems on targets which use the MPU for other purposes. Porter must make a decision to either support the MPU API by adding `MPU` to `device has`, or to not support the MPU API by defining `MBED_MPU_CUSTOM` so the MPU is free for customized use.
2. The information in targets.json needs to show what is actually supported.

closed 2019-03-12 by jeromecoutant