ARMmbed/mbed-os #9857: 802.15.4_RF can't compile is DEVICE_INTERRUPTIN is not enabled

jeromecoutant, 2019-02-26:
### Description

Hi

If you remove INTERRUPTIN from DEVICE_HAS json configuration, 
mbed-os compilation is failing:

````
$ mbed compile -t ARM -m NUCLEO_L476RG -v

Compile [  1.0%]: NanostackRfPhyMcr20a.cpp                                                                                                                                                                                        
[Error] NanostackRfPhyMcr20a.h@64,0:  #135: namespace "mbed"  has no member "InterruptIn"
[Error] NanostackRfPhyMcr20a.cpp@137,0:  #20: identifier "InterruptIn" is undefined
````

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-02-28:
Hi

Note that this issue is raised during TFM scope. (#9221) (#9853)

cc @ARMmbed/mbed-os-hal @ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-psa

0xc0170, 2019-02-28:
> 
[Error] NanostackRfPhyMcr20a.h@64,0:  #135: namespace "mbed"  has no member "InterruptIn"
[Error] NanostackRfPhyMcr20a.cpp@137,0:  #20: identifier "InterruptIn" is undefined

This is similar to usticker. Such basic functionality like InterruptIn would not be ported? As this option is there should be supported

@ARMmbed/mbed-os-ipcore

mikisch81, 2019-03-12:
> [Error] NanostackRfPhyMcr20a.h@64,0: #135: namespace "mbed" has no member "InterruptIn"
> [Error] NanostackRfPhyMcr20a.cpp@137,0: #20: identifier "InterruptIn" is undefined
>
> This is similar to [...]

@ARMmbed/mbed-os-ipcore can you address this issue?

 @ARMmbed/mbed-os-maintainers

artokin, 2019-03-21:
Fixed in PR https://github.com/ARMmbed/mbed-os/pull/10142.

jeromecoutant, 2019-03-21:
@artokin 
This is not fully fixed. I added a comment in #10142

closed 2019-03-21 by ciarmcom

jeromecoutant, 2019-03-21:
@ciarmcom please re-open this issue as it is not fixed
Thx

jeromecoutant, 2019-03-21:
````
[DEBUG] Output: In file included from .\components\802.15.4_RF\mcr20a-rf-driver\source\NanostackRfPhyMcr20a.cpp:16:
[DEBUG] Output: ./components/802.15.4_RF/mcr20a-rf-driver/mcr20a-rf-driver\NanostackRfPhyMcr20a.h:64:11: error: no type named 'InterruptIn' in namespace 'mbed'
[DEBUG] Output:     mbed::InterruptIn _rf_irq;
[DEBUG] Output:     ~~~~~~^
[DEBUG] Output: 1 error generated.
````

artokin, 2019-03-21:
Thanks @jeromecoutant , I think I compiled all drivers without DEVICE_INTERRUPTIN but apparently something was not right in my environment.

artokin, 2019-03-21:
A fix PR is available in https://github.com/ARMmbed/mbed-os/pull/10185