ARMmbed/mbed-os #9865: Memory management problem

masoudr, 2019-02-26:
[*] Question
    [ ] Enhancement
    [ ] Bug
Hi,
I'm using STM32F407VG with 192KB of RAM. Actually, my main code is huge and I trimmed it. I'm trying to read a file then generate a base64 string from it and send it as post_request with a gsm modem. But I'm unable to configure the memory stack sizes and get `Stack overflow` or `Operator new out of memory`, and I'm not sure if the problem is because of my coding or my configuration. I also read [this blog](https://os.mbed.com/blog/entry/Tracking-memory-usage-with-Mbed-OS/).
Here is some of the actual code:

```
#include "mbed.h"
#include "platform\ATCmdParser.h"
#include "drivers\UARTSerial.h"

void print_memory_info()
{
    // allocate enough room for every thread's stack statistics
    int cnt = osThreadGetCount();
    mbed_stats_stack_t *stats = (mbed_stats_stack_t*) malloc(cnt * sizeof(mbed_stats_stack_t));
 
    cnt = mbed_stats_stack_get_each(stats, cnt);
    for (int i = 0; i < cnt; i++) {
        printf("Thread: 0x%lX, Stack size: %lu / %lu\r\n", stats[i].thread_id, stats[i].max_size, stats[i].reserved_size);
    }
    free(stats);
 
    // Grab the heap statistics
    mbed_stats_heap_t heap_stats;
    mbed_stats_heap_get(&heap_stats);
    printf("Heap size: %lu / %lu bytes\r\n", heap_stats.current_size, heap_stats.reserved_size);
}

bool upload_report_main(string file_path)
{
    bool result = false;
    if (!sd.file_exist(file_path.c_str()))
        return result;
    long size = sd.file_size(file_path.c_str());
    int b64_size = base64.encode_size(size);
    FILE *fp = sd.open_file(file_path.c_str(), "rb");
    wait_ms(10);
    printf("START 950: ");
    print_memory_info();
    unsigned char *file_buffer = new unsigned char[size]();
    unsigned char *file_buffer_base64 = new unsigned char[b64_size]();
    wait_ms(10);
    printf("955: ");
    print_memory_info();
    int bytesRead = 0;
    // read up to sizeof(buffer) bytes
    if ((bytesRead = fread(file_buffer, 1, size, fp)) > 0)
    {
        printf("\r\nsize: %d, b64_size: %d\r\n", size, b64_size);
        wait_ms(10);
        printf("\r\n961: ");
        print_memory_info();
        // generate b64 from buffer
        if(base64.encode(file_buffer_base64, b64_size, file_buffer, size))
        {
            delete[] file_buffer;
            wait_ms(10);
            printf("966: ");
            print_memory_info();
            connection.upload_report("http://89.32.251.109:9090/RP", file_buffer_base64, b64_size);
            printf("982\r\n");
            delete[] file_buffer_base64;
            printf("984\r\n");   
            result = true;
        }
        else
        {
            delete[] file_buffer_base64;
[... 238 lines cut ...]
> 00:26:12.298 -> CPUID: 410FC241
> 00:26:12.298 -> HFSR : 40000000
> 00:26:12.298 -> MMFSR: 00000082
> 00:26:12.298 -> BFSR : 00000000
> 00:26:12.298 -> UFSR : 00000000
> 00:26:12.298 -> DFSR : 00000008
> 00:26:12.298 -> AFSR : 00000000
> 00:26:12.298 -> MMFAR: 0000000C
> 00:26:12.298 -> Mode : Thread
> 00:26:12.298 -> Priv : Privileged
> 00:26:12.298 -> Stack: PSP
> 00:26:12.298 -> 
> 00:26:12.298 -> -- MbedOS Fault Handler --
> 00:26:12.298 -> 
> 00:26:12.298 -> 
> 00:26:12.298 -> 
> 00:26:12.298 -> ++ MbedOS Error Info ++
> 00:26:12.333 -> Error Status: 0x80FF013D Code: 317 Module: 255
> 00:26:12.333 -> Error Message: Fault exception
> 00:26:12.333 -> Location: 0x800A1AF
> 00:26:12.333 -> Error Value: 0x8027F74
> 00:26:12.333 -> Current Thread: main  Id: 0x2000CDAC Entry: 0x801A1DB StackSize: 0x5000 StackMem: 0x20007D78 SP: 0x2001FF28 
> 00:26:12.333 -> For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF013D
> 00:26:12.333 -> -- MbedOS Error Info --
```

adbridge, 2019-02-27:
@masoudr Please do not delete any parts of the issue template when filling in your issues. Issue headers are parsed automatically and expect to see the full template. I will fix this for you this time.

jeromecoutant, 2019-03-05:
@naveenkaje why did you set device st label ?
I don't undertstand in the raised question that there is an issue ST side ?

vidavidorra, 2019-03-28:
> I'm using STM32F407VG with 192KB of RAM

Actually, you don't! That MCU has 128 kB of RAM and 64 kB of CCM, see the product page! Those are separate thinks. Maybe that explains your memory issues if you use more than that. Full disclaimer, I haven'r read your code so I don't know if you're already below that limit just thought to inform you on the usage.

See #8220 for using the CCM section.

Hope this helps

masoudr, 2019-03-28:
@vidavidorra Actually I solved the problem somehow but I was wondering why I get 128KB of memory instead of 192KB I tough that maybe `mbed-os` is occupying 64KB of memory, and of course, [this model](https://www.st.com/en/evaluation-tools/stm32f4discovery.html) has exact 192KB of SRAM.

closed 2019-03-28 by masoudr

vidavidorra, 2019-03-28:
> wondering why I get 128KB of memory instead of 192KB I tough that maybe mbed-os is occupying 64KB of memory

No, that is what I tried to explain in the previous comment. From that 192 kB, you get 128 kB of actual RAM and the other 64 kB is CCM. Mbed OS, by default, is only able to use the 128 kB RAM 'normally'. See the link in my previous comment on how the CCM can be used. It can **not** be added as normal RAM.

<br>

> and of course, this model has exact 192KB of SRAM.

In some way, note below how they specify this.

> Up to 192+4 Kbytes of SRAM including 64-Kbyte of CCM (core coupled memory) data RAM - Memories section on the [STM product page](https://www.st.com/en/microcontrollers-microprocessors/stm32f407vg.html)