ARMmbed/mbed-os #9868: TLS handshake failed due to the un-sorted server cert chain

ArthurMa1978, 2019-02-27:
### Description

The TLS handshake failed when trying to validate the cert chain from server side.
There have 4 certs return from server like bellowing:
```
   version: 3, type: 2340, ca 0, pathLen 0
   issuer:  App
   subject: Machine

   version: 3, type: 260, ca 1, pathLen 0
   issuer:  Root
   subject: Intermediate

   version: 3, type: 260, ca 1, pathLen 0
   issuer:  Intermediate
   subject: Machine

   version: 3, type: 260, ca 1, pathLen 1
   issuer:  Machine
   subject: App
```
The first cert is right which is the end-entity certificate, but next 3 are not ordered. But according to the [TLS 1.3](https://tools.ietf.org/id/draft-ietf-tls-tls13-23.html#certificate) this is acceptable, but [TLS 1.2](https://tools.ietf.org/html/rfc5246#section-7.4.2) force the order.
Since mbedTLS doesn't support TLS1.3, so the question is:
1. When will mbedTLS support TLS 1.3? There already have many same asks, so just raise it again.
2. Even without TLS 1.3 support, can have a fix to support this kind of scenario?

BTW, for other TLS providers which support TLS 1.3 can work properly with this un-sorted server cert chain.

### Issue request type

    [ ] Question
    [x] Enhancement
    [x] Bug

0xc0170, 2019-04-04:
Based on my review in our internal ticket and PR - not yet

0xc0170, 2019-08-14:
@Patater When this fix lands on master?