ARMmbed/mbed-os #9869: Gattserver crashes during discovery when more than 20 writable characteristics added

akkishore1, 2019-02-27:
When more than 20 writable characteristics are added then while adding  21st characteristic  it is returning from following code in function insert_characteristic_value_attribute().

if ( _auth_char_count >= MAX_CHARACTERISTIC_AUTHORIZATION_CNT) {
			printf(" BLE_ERROR_NO_MEM\n");
            return BLE_ERROR_NO_MEM;
        }

This error is not getting handled and the subsequent characteristics are getting added with same attribute_it. 

This error causes crash during discovery in the following function attsFindServiceGroupEnd().

This issue is found while porting homekit adk and issue is resolved by incrementing MAX_CHARACTERISTIC_AUTHORIZATION_CNT to 25 from 20.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

desmond-blue, 2019-03-04:
#9921

akkishore1, 2019-03-06:
Hi,
I have looked at the fix. Returning error is fine but this will not solve the actual error we are facing. In Homekit ADK there are more than 20 writable characteristics. So we will not be able to add those characteristics with this fix. As a work around I have increased MAX_CHARACTERISTIC_AUTHORIZATION_CNT to 25 from 20.

Also looking at the code I think there is one more bug in the code which actually causing the issue even though characteristics are only writable characteristics and not authorization characteristics. 
I think there is overlap between bitfields of  UPDATE_PROPERTIES which is BLE_GATT_CHAR_PROPERTIES_NOTIFY | BLE_GATT_CHAR_PROPERTIES_INDICATE which is 0x30 and ATTS_PERMIT_WRITE which is 0x10. I think this is what is causing it to enter the below condition in the first place.

if ( _auth_char_count >= MAX_CHARACTERISTIC_AUTHORIZATION_CNT) {
printf(" BLE_ERROR_NO_MEM\n");
return BLE_ERROR_NO_MEM;
}

desmond-blue, 2019-03-07:
@pan- Would you help on this, why does [UPDATE_PROPERTIES be checked](https://github.com/ARMmbed/mbed-os/blob/f8d254fc4c4a9b41351499861770c58bc00b86d0/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioGattServer.cpp#L388) for registering authorisation list?

pan-, 2019-03-07:
@paul-szczepanek-arm Could you introduce a new configuration parameter for `MAX_CHARACTERISTIC_AUTHORIZATION_CNT` ?

paul-szczepanek-arm, 2019-03-07:
Sure. IOTPAN-405

akkishore1, 2019-04-05:
Fixed in Master branch of mbed. Verified the same

closed 2019-04-05 by akkishore1