ARMmbed/mbed-os #9877: suggestion: separate targets floder into a submodule

whutddk, 2019-02-27:
### Description

The repo is so big that it always takes too much time to clone it. Only one of the targets in repo needed to used by one users usually. Why not set up a submodule to store target files and fetch them automatically with a script?

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

40Grit, 2019-02-28:
Try mirroring the repo elsewhere or to a local git server. For some reason cloning it from gitlab is unbelievably slow.

40Grit, 2019-02-28:
As an additional note I believe Peru might also allow you to clone only particular sub-folders. Such that you could pull down only the targets you want.

bulislaw, 2019-05-07:
At this stage we are not considering using submodules, there are ongoing discussion about how to include external dependencies though. We won't be able to address it anytime soon, therefore closing the issue.

closed 2019-05-07 by bulislaw

loverdeg-ep, 2019-05-07:
Final note on this. The latest version of [Buckaroo](https://buckaroo.pm/) looks promising.