ARMmbed/mbed-os #9880: SecureStore and Cell with nRF52840

trowbridgec, 2019-02-27:
### Description
Target: nRF52840 on a custom board with onboard cell modem (Telit ME910)
Toolchain: GCC_ARM
Mbed OS version: Currently 5.11.0

In our efforts to get cell working (Telit ME910), we found that we had to disable `CRYPTOCELL310` and `MBEDTLS_CONFIG_HW_SUPPORT` in our `custom_targets.json` file to work around issues with the underlying NRF52 SDK 14.2 (which doesn't support Cryptocell, or so I believe?). See [this fork](https://github.com/karsev/mbed-os-example-cellular) of the `mbed-os-example-cellular` project which uses the same target.

For our application, we'd also like to utilize the SecureStore functionality, but this requires mbedTLS to be configured. What steps can we take to initialize mbedTLS with Cryptocell and mbedTLS HW config disabled?

Related issues:
* #8666 

@dlfryar 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

TacoGrandeTX, 2019-02-28:
@trowbridgec We currently use the CC310 for BLE with the NRF52840 so some functionality is there. I was able to build the fork you referenced with ```CRYPTOCELL310``` and ```MBEDTLS_CONFIG_HW_SUPPORT``` enabled but don't have suitable hardware for testing.  Are you seeing run-time or build issues?

@RonEld - any thoughts on this?

TacoGrandeTX, 2019-03-01:
I was able to build the forked mbed-os-example-cellular project with ```CRYPTOCELL310``` and ```MBEDTLS_CONFIG_HW_SUPPORT```, but I don't have the required hardware to run the example.

loverdeg-ep, 2019-03-01:
@TacoGrandeTX the fork is pinned to mbed-os master.

We will be updating to 5.11.5 shortly but are you able to try again with mbed-os 5.11.0

trowbridgec, 2019-03-01:
@TacoGrandeTX So far, I've traced the issue with `CRYPTOCELL310`, `MBEDTLS_CONFIG_HW_SUPPORT`, and cell to a run time issue with the LWIP stack. From my investigations so far, it looks like something to do with the LWIP constructor needing to generate some random numbers? It gets to a certain point and then stops responding.

I'll update if I find out more; I'm currently trying to follow down the trail to figure out what the real issue is.

Note: I also updated to `mbed-os-5.11.5`, but it didn't seem to help fix anything.

trowbridgec, 2019-03-01:
More info: I tracked the issue down further to [`mbed-os/features/cryptocell/FEATURE_CRYPTOCELL310/trng.c`, line 69](https://github.com/ARMmbed/mbed-os/blob/master/features/cryptocell/FEATURE_CRYPTOCELL310/trng.c#L69); the `LLF_RND_GetTrngSource()` function.

I found this [issue](https://github.com/openthread/openthread/issues/2898) on the OpenThread repo which seems like it might be related?

trowbridgec, 2019-03-01:
After reviewing a couple issues related to `CC310` and `nRF52840`, I came across #7099 and tried adding a call to `mbedtls_platform_setup()` before starting my cell connection, and it seems to have fixed the issue I was seeing.

My ultimate goal here is to run the [simple-mbed-cloud-client](https://github.com/ARMmbed/simple-mbed-cloud-client), and addressing this issue was the first hurdle. Now, I'm getting errors relating to the ESFS:
```
[INFO][esfs]: esfs_init - enter
[ERR ][PAL ]: Failed to create folder, was the storage properly initialized?
[ERR ][esfs]: esfs_init() 0
[ERR ][PAL ]: Failed to create folder, was the storage properly initialized?
[ERR ][esfs]: esfs_init() 1
```
But, I think that might be a topic for a separate issue.

I will still try to run the SecureStore test now that we have `CC310` enabled, and see if it passes.

RonEld, 2019-03-03:
@trowbridgec I appologize for not replying, as I was away.
Yes, to initialize platform for Mbed TLS, the functions `mbedtls_platform_setup()` and `mbedtls_platform_terminate()` should becalled.
It currently affects only NRF 52840_DK.
Has https://github.com/ARMmbed/mbed-os/pull/9493 fixed your issue?

loverdeg-ep, 2019-03-04:
> It currently affects only NRF 52840_DK.

@RonEld Thanks! We have designed a custom target. Does the issue affect the NRF52840 as an MCU or the NRF52840-DK? Any particulars on why?

RonEld, 2019-03-05:
@loverdeg-ep 

> Does the issue affect the NRF52840 as an MCU or the NRF52840-DK? Any particulars on why?

The reason is because `mbedtls_platform_setup()` and `mbedtls_platform_terminate()` was designed to initalize the platform needed for Mbed TLS. First and foremost, the hardware acceleration engines, where needed. On platforms using Cryptocell310, NRF52840  being one of them, these functions initialize the Cryptocell clocks, so not calling these functions ( or the internal functions being called in the [implementation](https://github.com/ARMmbed/mbed-os/blob/master/features/cryptocell/FEATURE_CRYPTOCELL310/TARGET_MCU_NRF52840/crypto_device_platform.c#L28)) will cause the calls to cryptographic functionality to be stuck waiting for interrupts from the hardware that wasn't initialized.
Other targets at the moment haven't implemented these functions, as their hardware acceleration is independent for each cryptographic module.

loverdeg-ep, 2019-03-05:
@RonEld @dlfryar We've been targeting specific releases assuming it would be best to start our efforts on a non-moving target.

Should we start working a little closer to master?

RonEld, 2019-06-04:
@loverdeg-ep Depends on what version you are using.
I recommend you update your version closer to master, at least branch `mbed-os-5.11`

RonEld, 2019-06-04:
Do you still reproduce the issue after you call `mbedtls_platform_setup()` and `mbedtls_platform_terminate()`?

trowbridgec, 2019-06-04:
@RonEld After adding the calls to `mbedtls_platform_setup()` and `mbedtls_platform_terminate()` in our application we're now able to leave `CRYPTOCELL310` and `MBEDTLS_CONFIG_HW_SUPPORT` enabled.

Thanks for your help!

closed 2019-06-04 by trowbridgec