ARMmbed/mbed-os #9881: Is SDIO host mode enabled?

andrewc-arm, 2019-02-28:
### Description

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

Hi,

My customer is trying to enable SDIO host mode.
Does mbed-os support SDIO host mode?
If so, do we have example about it?

Thanks,

0xc0170, 2019-02-28:
The only reference to sdio I know is https://github.com/ARMmbed/sdio-driver/pull/1 (not completed yet). @ARMmbed/mbed-os-storage team might know more

andrewc-arm, 2019-03-04:
Hi, @ARMmbed/mbed-os-storage
Any chance that I can have your feedback? 😃

dannybenor, 2019-03-04:
@andrewc-arm The SDIO HAL does not exist, therefore the SDIO blockdevice was moved to an external repo. We do not test this blockdevice, that may work for the customer. Mbed OS currently does not support SDIO

andrewc-arm, 2019-03-04:
Thanks for the info, @dannybenor. We will think about what to do next.

closed 2019-03-04 by andrewc-arm