ARMmbed/mbed-os #9887: LittleFS: Iterating on a directory with dir_read while removing files produces phantom files

davidsaada, 2019-02-28:
### Description
In LittleFS, when iterating on a directory and removing all underlying files, The `dir_read` function produces non existing files. This is pretty rare, but reproducible. 
Steps to reproduce: Following code creates a LittleFS instance on a heap block device and then creates a directory on top. Then it creates 25 randomly named files (but with a fixed prefix) in this directory and fills them with some random data. After that, it iterates on all files in the directory and deletes them. When reading the directory, file names are tested for the prefix, where failing the check stops the test. The whole process is repeated 50 times.
Now, the problematic scenario occurs in the 22th iteration, where line 67 asserts for a wrong file name. Debug prints usually show an erroneous name, filled with the 0xFF character.
Code is compiled with GCC_ARM and executed with mbed test on the K64F board.

```
#include "HeapBlockDevice.h"
#include "LittleFileSystem.h"
#include "Dir.h"
#include "File.h"
#include "greentea-client/test_env.h"
#include "unity.h"
#include "utest.h"
#include <stdio.h>
#include <string.h>

using namespace utest::v1;
using namespace mbed;

void test_file_system_remove_while_iterating()
{
    BlockDevice *bd = new HeapBlockDevice(256 * 512, 512);
    TEST_ASSERT_NOT_EQUAL(0, bd);

    int err;

    err = LittleFileSystem::format(bd);
    TEST_ASSERT_EQUAL(0, err);

    FileSystem *fs = new LittleFileSystem("lfs");

    err = fs->mount(bd);
    TEST_ASSERT_EQUAL(0, err);

    File file;
    const char* prefix = "file_name_";
    const char* dir_name = "/test";
    char full_path[128];
    Dir dir;
    struct dirent dir_ent;

    err = fs->mkdir(dir_name, 0777);
    TEST_ASSERT_EQUAL(0, err);

    for (int i = 0; i < 50; i++) {
        printf("running iteration %d\n", i);
        for (int j = 0; j < 24; j++) {
            uint8_t value[256];
            srand(i * 50 + j);

            sprintf(full_path, "%s/%s%d", dir_name, prefix, rand() % 1000);
            err = file.open(fs, full_path, O_WRONLY | O_CREAT | O_TRUNC);
            TEST_ASSERT_EQUAL(0, err);

            int s = 1 + rand() % (sizeof(value) - 1);
            memset(value, (i * 50 + j) % 256, s);

            err = file.write(value, s);
            TEST_ASSERT_EQUAL(s, err);

            err = file.close();
            TEST_ASSERT_EQUAL(0, err);
        }

        err = dir.open(fs, dir_name);
        TEST_ASSERT_EQUAL(0, err);

        while (dir.read(&dir_ent) != 0) {
            if (!strcmp(dir_ent.d_name, ".") || !strcmp(dir_ent.d_name, "..")) {
                continue;
            }

            TEST_ASSERT_EQUAL(0, strncmp(dir_ent.d_name, prefix, strlen(prefix)));

            sprintf(full_path, "%s/%s", dir_name, dir_ent.d_name);

            err = fs->remove(full_path);
            TEST_ASSERT_EQUAL(0, err);
        }

        err = dir.close();
        TEST_ASSERT_EQUAL(0, err);
    }

    err = fs->unmount();
    TEST_ASSERT_EQUAL(0, err);

    delete bd;
}

utest::v1::status_t test_setup(const size_t number_of_cases)
{
    GREENTEA_SETUP(120, "default_auto");
    return verbose_test_setup_handler(number_of_cases);
}

Case cases[] = {
    Case("Remove while iterating", test_file_system_remove_while_iterating)
};

Specification specification(test_setup, cases);

int main()
{
    return !Harness::run(specification);
}

```

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

davidsaada, 2019-02-28:
@geky @dannybenor

dannybenor, 2019-02-28:
@davidsaada great investigation work!

closed 2020-05-11 by ciarmcom