ARMmbed/mbed-os #9890: i2c_t object not initialized to 0 causes timeout error

mrcoulter45, 2019-02-28:
@jeromecoutant 
https://github.com/ARMmbed/mbed-os/blob/0e7f112653b6d191e793f5ff30c755397b7f0275/targets/TARGET_STM/i2c_api.c#L330
When calling i2c_init(i2c_t *obj, PinName sda, PinName scl) in i2c_api.c, it is possible that obj_s->hz could be set to some random value. If that value is greater than SystemCoreClock (line 69), BYTE_TIMEOUT will be set to 0. When line 210 runs in i2c_hw_reset, timeout will be set to BYTE_TIMEOUT, and timeout will roll over to 0xffffffff in line 211 because of the "--timeout != 0" part of the while condition. Memsetting the object to 0 in my main program fixed the issue.

LMESTM, 2019-02-28:
@mrcoulter45 thanks for reporting this.

Isn't the i2c_t objects parameters supposed to be set by by the C++ driver so that it will never be a random value ? 
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.cpp#L37

The HAL layer is only supposed to be a wrapper layer to mbed-os drivers and makes assumption on what is taken care of by upper layers ...

But in any case, I guess we could complement the check with something like:
`if ((!obj_s->hz) || (obj_s-Hz > SystemCoreClock) {`

Would you feel like sending a PR  for the issue you've pointed to and investigated so far ?

mrcoulter45, 2019-02-28:
Yes the C++ drivers definitely make this safe, however, according to the porting docs, it should be safe to call i2c_init without things breaking. I can definitely put up a PR for a fix.

LMESTM, 2019-02-28:
> I can definitely put up a PR for a fix.

Would be much appreciated - we're definitely welcoming all contributions and you've done the most difficult part. We will be able to start automated testing as well on all of your target families.

mrcoulter45, 2019-02-28:
Raised PR #9895

mrcoulter45, 2019-03-20:
PR for fix has been merged into master.

closed 2019-03-20 by mrcoulter45

LMESTM, 2019-03-20:
thanks for this @mrcoulter45

mrcoulter45, 2019-03-20:
Absolutely