ARMmbed/mbed-os #9897: Cypress PSOC6 Targets missing QSPI Mbed HAL implementation

maclobdell, 2019-03-01:
### Description

The CY8CKIT-062-WiFi-BT board has a S25FL512SAGBHIA10 Quad SPI Flash chip which supports SFDP.  This device seems perfect for the Mbed QSPI HAL.  Adding an implementation for the QSPI HAL on PSOC6 targets with a Quad SPI would enable Little File System and Fat File System, and more sophisticated applications.  

#### Details 
Mbed OS File Systems run on top of a Block Device API as referenced here https://os.mbed.com/docs/mbed-os/v5.11/reference/storage-tech.html and https://os.mbed.com/docs/mbed-os/v5.11/apis/blockdevice.html
Mbed OS has a QSPIF Block Device driver https://os.mbed.com/docs/mbed-os/v5.11/apis/qspifblockdevice.html
If QSPI Mbed HAL api is implemented for a target, as referenced here https://os.mbed.com/docs/mbed-os/v5.11/porting/quadspi-port.html, then the QSPI Block Device Driver should just work.
Alternatively, a custom block device driver could be implemented if the above is not possible.  

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

maclobdell, 2019-03-01:
cc @ARMmbed/team-cypress @orenc17 @kfnta @dannybenor

maclobdell, 2019-03-08:
@Team-Cypress - any ETA on this? it would be much appreciated

johndebudge, 2019-03-08:
@maclobdell we do not have an exact ETA at this time but it is on our backlog of work to do.

maclobdell, 2019-04-17:
@johndebudge - any update on this?  will the integration point be qspifblockdevice or is your team planning a new smif blockdevice class?

0xc0170, 2019-04-18:
https://github.com/ARMmbed/mbed-os/pull/10435

maclobdell, 2019-05-08:
fix merged.  closing

closed 2019-05-08 by maclobdell