ARMmbed/mbed-os #9901: NRF52 DK Cordio BLE with button 1 or 3 interupt in

danielknox, 2019-03-01:
### Description

We are using mbed-os-5.11.4, developing with the Mbed online compiler. We find the following problem when we enable the BLE stack on the NRF52 DK (n52832). 

Simple example here:
https://os.mbed.com/teams/The-Shed/code/nRF52_Button_MWE/file/628e1249eaa1/src/main.cpp/

Our mbed_app.json contains:
`
{
    "target_overrides": {
        "NRF52_DK": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL", "NORDIC_SOFTDEVICE"]
        }
    }
}
`

When using the mbed_app.json file:

BLE enabled, interrupts enabled to button 1 or Button 3 fail to fire (interrupts should schedule task on an event queue to output some text to the serial port). BLE enabled, with interrupts declared to button 1 and button 3 causes the board to fail to run the program. BLE disabled, interrupts to button 1 or button 3 run fine. Interrupts associated to Buttons 2 and 4 work fine. 

Removing mbed_app.json:
Interrupts work fine, BLE now doesn't work.

Does anyone know why enabling BLE is causing a conflict with these particular pins?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

desmond-blue, 2019-03-04:
Hi @paul-szczepanek-arm ,

Per my understanding, Cordio is taking control of some gpios for LEDs, and the pins are different between NRF52_DK and NRF52840_DK, so if libcordio_stack_nordic.a is built with platform NRF52840_DK, the pins may conflict.

[Pin define of NRF52_DK](https://github.com/ARMmbed/mbed-os/blob/fa65546aa2b165696b6b99497010a0a07d33e962/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/TARGET_NRF52_DK/PinNames.h#L125)
```
    LED1    = p17,
    LED2    = p18,
    LED3    = p19,
    LED4    = p20,

    BUTTON1 = p13,
    BUTTON2 = p14,
    BUTTON3 = p15,
    BUTTON4 = p16,
```

[Pin define of NRF52840_DK](https://github.com/ARMmbed/mbed-os/blob/fa65546aa2b165696b6b99497010a0a07d33e962/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/PinNames.h#L163)
```
    LED1    = p13,
    LED2    = p14,
    LED3    = p15,
    LED4    = p16,

    BUTTON1 = p11,
    BUTTON2 = p12,
    BUTTON3 = p24,
    BUTTON4 = p25,
```

marcemmers, 2019-03-28:
We are having similar problems where the Cordio layer is taking over some of our pins. By coincidence those pins are also leds on our custom board causing them to flicker. Probably some of the other pins are also causing problems.

Why is the cordio stack controlling pins in the first place? Is there a way to disable it in the cordio stack?

paul-szczepanek-arm, 2019-04-17:
this is fixed in the 19.2 cordio LL, PR pending preceding PR

desmond-blue, 2019-07-24:
Hi @danielknox, this should be fixed in mbed-os.5.13.0 by #10666 , would you be able to confirm it and close this issue? Thanks.

closed 2019-08-14 by 0xc0170