ARMmbed/mbed-os #9912: FRDM k64F devihas

NiranjanRavi1993, 2019-03-03:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-03-07:
It's not just adding one macro. The MCU family does not have CAN implementation

cc @ARMmbed/team-nxp

niranjanfall2017, 2019-03-09:
Hi,
    The MCU family has CAN pins available. They do not have trans-receiver component needed to complete the operation. I was able to use and implement it in **MCUxpresso** by adding suitable trans-receive. I wanted to give it a try in mbed platform because the code is easily understandable.

NiranjanRavi1993, 2019-03-16:
hi @ARMmbed/team-nxp
        It would be great if someone could help me with this. Thanks

NiranjanRavi1993, 2019-03-31:
Hi,
   I am not sure whether this issue is under consideration if not i can close this issue. 
Regards
Niranjan

cmonr, 2019-04-01:
@ARMmbed/team-nxp @maclobdell

NiranjanRavi1993, 2019-04-05:
Hi @maclobdell 
can you please have a look. It would be really helpful.
Thanks and Regards
Niranjan

NiranjanRavi1993, 2019-06-24:
Closing this cause no updates were received.

closed 2019-06-24 by NiranjanRavi1993