ARMmbed/mbed-os #9922: Compilation Error on Latest mbed-os

mudassar-ublox, 2019-03-04:
### Description
Getting compilation error in LWIPStack class on latest mbed-os. Error occurs when "lwip.tcp-enabled" is disable  in mbed_app.json.
            "lwip.tcp-enabled"                  : false,

[Error] LWIPStack.cpp@519,33: invalid use of incomplete type 'struct tcp_pcb'

[Mbed_Build.txt](https://github.com/ARMmbed/mbed-os/files/2924495/Mbed_Build.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

fahim-ublox, 2019-03-04:
@RobMeades @screamerbg

mudassar-ublox, 2019-03-04:
Getting error after this commit on 21 Feb 2019
[Multihoming initial release (#9387)](https://github.com/ARMmbed/mbed-os/commit/dbd92c746bb2ae722f8f5301d2fd6ca9fa28a500 )

SeppoTakalo, 2019-03-08:
Thanks for the report,
I'll assign this to be fixed within my team.

SeppoTakalo, 2019-03-26:
Fixed in https://github.com/ARMmbed/mbed-os/pull/10190

closed 2019-03-26 by ciarmcom