ARMmbed/mbed-os #9933: Green tea test failing on EFM32GG11_STK3701

deepikabhavnani, 2019-03-04:
### Description
```
mbedgt: test suite report:
| target                | platform_name     | test suite                                       | result | elapsed_time (sec) | copy_method |
|-----------------------|-------------------|--------------------------------------------------|--------|--------------------|-------------|
| EFM32GG11_STK3701-ARM | EFM32GG11_STK3701 | mbed-os-tests-mbed_hal-stack_size_unification    | FAIL   | 21.14              | default     |
| EFM32GG11_STK3701-ARM | EFM32GG11_STK3701 | mbed-os-tests-mbedmicro-rtos-mbed-heap_and_stack | FAIL   | 21.5               | default     |

```
### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-03-08:
cc @ARMmbed/team-silabs

stevew817, 2019-03-08:
We didn't submit any patches to EFM32 lately, so I'm taking a guess this is related to #9571.

I can't see the test job results for #9571. If I had to guess, these didn't run against EFM32 and therefore caused failure down the line when EFM32GG11_STK3701 is now being tested as part of pelion_enablement.

@0xc0170 Can you check whether #9571 was run against EFM32 boards before being merged in, before I spend any time looking into this?

0xc0170, 2019-03-08:
> @0xc0170 Can you check whether #9571 was run against EFM32 boards before being merged in, before I spend any time looking into this?

Reviewed. It did not run for the PR.

stevew817, 2019-03-08:
@deepikabhavnani I guess it's back to you then, since you're the author of that PR.

stevew817, 2019-03-08:
On a broader note, @0xc0170 do you have any idea *why* it wasn't run?

stevew817, 2019-03-08:
@mprse Can you take a look at this, since the apparently offending PR is yours?

0xc0170, 2019-03-08:
We have nightly running - has wider coverage. The latests runs do not contain emf32. I'll talk to the test team

cc @ARMmbed/mbed-os-test

deepikabhavnani, 2019-03-08:
@stevew817 - issue was identified and fix was going as part of other PR, this was to record that we get the fix in 5.12.
#9910 has added fix for this. https://github.com/ARMmbed/mbed-os/pull/9910/commits/8b25887021c58261204935083ab477b467e08c3d
Sorry for trouble. Will close this once PR is merged

deepikabhavnani, 2019-03-08:
Also though device was not part of CI test were executed locally to verify linker script updates

cmonr, 2019-03-12:
Closing, since PR is merged.

closed 2019-03-12 by cmonr