ARMmbed/mbed-os #9934: NUCLEO_F767ZI FLASHIAP erase functionality error

theamirocohen, 2019-03-05:
### Description

Following https://jira.arm.com/browse/IOTSTOR-793 tickect, I've noticed some erase issues with the following device (the device only storage component is flashiap).
While running features-storage-tests-blockdevice-general_block_device tests, the test_erase_functionality fails, the test flow is as follows:
1. erase a sector.
2. write to the sector random data.
3. read and verify the data.
4. erase the sector again.
5. verify that the sector was erased. <== FAIL.

After printing the sector data I've noticed that the sector was partly erased (in arbitrary places).
The fail happen on all toolchains (ARM, IAR, GCC).

This was also reproduced in tests-mbed_drivers-flashiap tests, the flashiap_program_test randomly fails.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

@dannybenor 
@deepikabhavnani

NirSonnenschein, 2019-03-05:
assigned to @jeromecoutant  at @ARMmbed/mbed-os-storage 's request

VVESTM, 2019-03-06:
Is it possible to have more details about this IOTSTOR-793 ticket ? the link is not accessible... Is it a regression ?

Then, do you have a main.c that can be used to reproduce this issue ?

theamirocohen, 2019-03-07:
> Is it possible to have more details about this IOTSTOR-793 ticket ? the link is not accessible... Is it a regression ?
> 
> Then, do you have a main.c that can be used to reproduce this issue ?

Hi, it's not regression, the board was added into MbedOS CI nightly tests and failed the current test.
IOTSTOR-793 has no information about the issue, just the task of examining the device's fails.
A main.c can be found [here](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/TESTS/blockdevice/general_block_device/main.cpp), the function that fails is test_erase_functionality, at the erase verification step.
If you need more info please let me know.

VVESTM, 2019-03-13:
Hi, I can reproduce this test with STM32 Cube HAL.
There is also an other ticket related to this issue : https://github.com/ARMmbed/mbed-os/issues/6380

The issue is probably due to a cache issue. Analyse on going.
Waiting for confirmation, you can call SCB_InvalidateDCache() just after the erase.

offirko, 2019-03-14:
@VVESTM - could this issue be related to: https://github.com/ARMmbed/mbed-os/issues/10049 ?

VVESTM, 2019-03-15:
@offirko, I can reproduce #10049 even when calling SCB_InvalidateDCache() at end of erase.
So must be an other issue...

kjbracey-arm, 2019-03-26:
Global `SCB_InvalidateDCache` is not something you want to do - you don't want to forget any data in the cache that's waiting to be written back to RAM.

But looking at flash_api.c, I guess you do want a ranged `SCB_InvalidateDCache` covering the area you just erased or programmed (both cases), and a (global) `SCB_InvalidateICache`.

(You only really need to invalidate after erasing if you're going to insist on reading back the erased values and check them, which you are - otherwise conceivably you could just invalidate after programming).

PR #10248 (2019-04-01): TARGET_STM32F7: Refresh cache when erasing or programming flash
TARGET_STM32F7: Refresh cache when erasing or programming flash
    
The cache must be refreshed when we erase or program flash memory.
It fix 2 issues :
    Fix #9934
    Fix #6380
    
This solution was initially proposed in #6380.

Solution tested on mbed-os-5.12.0 and STM32F746 Disco board with this command: 
    mbed test -t ARM -m DISCO_F746NG -v -n *features-storage-tests-blockdevice-general_block_device* -v
commit 5765c4d0 (2019-04-01): TARGET_STM32F7: Refresh cache when erasing or programming flash

The cache must be refreshed when we erase or program flash memory.
It fix 2 issues :
    Fix #9934
    Fix #6380

This solution was initially proposed in #6380.
closed 2019-04-01 by cmonr