ARMmbed/mbed-os #9938: make armc6: error 8 for assembly file

0xc0170, 2019-03-05:
### Description

Creating tracking issue for this as it seems no rootcase found so far.

Recent nightly and some PRs failed exporters job  (since yesterday as soon as make armc6 got actived in the exporters).

The failure indicates a problem with `mbed_config.h` file include in the assembly files. This is the log from CI:

```
==========STDERR==========
Warning: A3912W: Option 'c' is deprecated.
<built-in>:388:10: fatal error: 'mbed_config.h' file not found
#include "./mbed_config.h"
         ^~~~~~~~~~~~~~~~~
1 error generated.
Fatal error: A1905U: Pre-processor step failed for '../mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.S'
1 Error, 1 Warning
make[1]: *** [mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.o] Error 8
```

Interesting part, I was able to reproduce the same failure but only if I had armc5 in paths as well (and many other failures because of different armasm) - using windows. Our CI that is failing has only ARMC6 installed, it's linux based machine.

Locally, executed `python -u mbed-os/tools/test/examples/examples.py export make_armc6 --mcu K66F`

This is the error I get: 

```
process_begin: CreateProcess(NULL, C:/Program Files/Git/bin/sh.exe -c "echo \"main.o mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_ARM/except.o mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o mbed-os/cmsis/TARGET_CORTEX_M/mbed_tz_context.o mbed-os/components/802.15.4_RF/atmel-rf-driver/source/NanostackRfPhyAtmel.o mbed-os/components/802.15.4_RF/atmel-rf-driver/source/at24mac.o mbed-os/components/802.15.4_RF/mcr20a-rf-driver/source/MCR20Drv.o mbed-os/components/802.15.4_RF/mcr20a-rf-driver/source/NanostackRfPhyMcr20a.o mbed-os/components/802.15.4_RF/stm-s2lp-rf-driver/source/NanostackRfPhys2lp.o mbed-os/components/802.15.4_RF/stm-s2lp-rf-driver/source/rf_configuration.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_EMUL/psa_attest_inject_key.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_EMUL/psa_initial_attestation_api.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/attest_boot_status_loader.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/attest_crypto.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/attest_crypto_keys.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/attest_iat_claims_loader.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/attestation_bootloader_data.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/psa_attestation_stubs.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/psa_inject_attestation_key_impl.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/tfm_impl/attest_token.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/tfm_impl/attestation_core.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/tfm_impl/t_cose/src/t_cose_sign1_sign.o mbed-os/components/TARGET_PSA/services/attestation/COMPONENT_PSA_SRV_IMPL/tfm_impl/t_cose/src/t_cose_util.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/src/UsefulBuf.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/src/ieee754.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/src/qcbor_decode.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/src/qcbor_encode.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/UsefulBuf_Tests.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/float_tests.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/half_to_double_from_rfc7049.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/qcbor_decode_tests.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/qcbor_encode_tests.o mbed-os/components/TARGET_PSA/services/attestation/qcbor/test/run_tests.o mbed-os/components/TARGET_PSA/services/platform/COMPONENT_PSA_SRV_EMUL/platform_emul.o mbed-os/components/TARGET_PSA/services/platform/COMPONENT_PSA_SRV_IMPL/platform_srv_impl.o mbed-os/components/TARGET_PSA/services/storage/common/psa_storage_common_impl.o mbed-os/components/TARGET_PSA/services/storage/its/COMPONENT_PSA_SRV_EMUL/psa_prot_internal_storage.o mbed-os/components/TARGET_PSA/services/storage/its/COMPONENT_PSA_SRV_IMPL/pits_impl.o mbed-os/components/TARGET_PSA/services/storage/ps/COMPONENT_NSPE/protected_storage.o mbed-os/components/storage/blockdevice/COMPONENT_FLASHIAP/COMMON/fslittle_test.o mbed-os/components/storage/blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.o mbed-os/components/storage/blockdevice/COMPONENT_SD/SDBlockDevice.o mbed-os/components/wifi/esp8266-driver/ESP8266/ESP8266.o mbed-os/components/wifi/esp8266-driver/ESP8266Interface.o mbed-os/drivers/AnalogIn.o mbed-os/drivers/BusIn.o mbed-os/drivers/BusInOut.o mbed-os/drivers/BusOut.o mbed-os/drivers/CAN.o mbed-os/drivers/Ethernet.o mbed-os/drivers/FlashIAP.o mbed-os/drivers/I2C.o mbed-os/drivers/I2CSlave.o mbed-os/drivers/InterruptIn.o mbed-os/drivers/InterruptManager.o mbed-os/drivers/MbedCRC.o mbed-os/drivers/QSPI.o mbed-os/drivers/RawSerial.o mbed-os/drivers/SPI.o mbed-os/drivers/SPISlave.o mbed-os/drivers/Serial.o mbed-os/drivers/SerialBase.o mbed-os/drivers/TableCRC.o mbed-os/drivers/Ticker.o mbed-os/drivers/Timeout.o mbed-os/drivers/Timer.o mbed-os/drivers/TimerEvent.o mbed-os/drivers/UARTSerial.o mbed-os/events/EventQueue.o mbed-os/events/equeue/equeue.o mbed-os/events/equeue/equeue_mbed.o mbed-os/events/equeue/equeue_posix.o mbed-os/events/mbed_shared_queues.o mbed-os/features/cellular/framework/AT/ATHandler.o mbed-os/features/cellular/framework/AT/ATHandler_factory.o mbed-os/features/cellular/framework/AT/AT_CellularBase.o mbed-os/features/cellular/framework/AT/AT_CellularContext.o mbed-os/features/cellular/framework/AT/AT_CellularDevice.o mbed-os/features/cellular/framework/AT/AT_CellularInformation.o mbed-os/features/cellular/framework/AT/AT_CellularNetwork.o mbed-os/features/cellular/framework/AT/AT_CellularSMS.o mbed-os/features/cellular/framework/AT/AT_CellularStack.o mbed-os/features/cellular/framework/AT/AT_ControlPlane_netif.o mbed-os/features/cellular/framework/common/CellularLog.o mbed-os/features/cellular/framework/common/CellularUtil.o mbed-os/features/cellular/framework/device/CellularContext.o mbed-os/features/cellular/framework/device/CellularDevice.o mbed-os/features/cellular/framework/device/CellularStateMachine.o mbed-os/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION.o mbed-os/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularContext.o mbed-os/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularInformation.o mbed-os/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularStack.o mbed-os/features/cellular/framework/targets/GENERIC/GENERIC_AT3GPP/GENERIC_AT3GPP.o mbed-os/features/cellular/framework/targets/MultiTech/DragonflyNano/PPP/SARA4_PPP.o mbed-os/features/cellular/framework/targets/MultiTech/DragonflyNano/PPP/SARA4_PPP_CellularNetwork.o mbed-os/features/cellular/framework/targets/QUECTEL/BC95/QUECTEL_BC95.o mbed-os/features/cellular/framework/targets/QUECTEL/BC95/QUECTEL_BC95_CellularContext.o mbed-os/features/cellular/framework/targets/QUECTEL/BC95/QUECTEL_BC95_CellularInformation.o mbed-os/features/cellular/framework/targets/QUECTEL/BC95/QUECTEL_BC95_CellularNetwork.o mbed-os/features/cellular/framework/targets/QUECTEL/BC95/QUECTEL_BC95_CellularStack.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_CellularContext.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_CellularInformation.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_CellularNetwork.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_CellularStack.o mbed-os/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_ControlPlane_netif.o mbed-os/features/cellular/framework/targets/QUECTEL/M26/QUECTEL_M26.o mbed-os/features/cellular/framework/targets/QUECTEL/M26/QUECTEL_M26_CellularContext.o mbed-os/features/cellular/framework/targets/QUECTEL/M26/QUECTEL_M26_CellularInformation.o mbed-os/features/cellular/framework/targets/QUECTEL/M26/QUECTEL_M26_CellularStack.o mbed-os/features/cellular/framework/targets/QUECTEL/UG96/QUECTEL_UG96.o mbed-os/features/cellular/framework/targe [... cut ...] ties/OperationListBase.o mbed-os/usb/device/utilities/events/PolledQueue.o mbed-os/usb/device/utilities/events/TaskBase.o\" > .link_options.txt", ...) failed.
make (e=87): The parameter is incorrect.
make[1]: *** [mbed-os-example-blinky.elf] Error 87
make: *** [makefile:26: all] Error 2
```

Same error if I run just `make` in the example blinky (using make 4.2 or 3.81)

Too long command?

@SenRamakri @ARMmbed/mbed-os-test 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-05:
One of the jobs affected: https://github.com/ARMmbed/mbed-os/pull/9738#issuecomment-469467096

ARMC6 landed without exporters test for armc6 make:  https://github.com/ARMmbed/mbed-os/pull/9888#issuecomment-468877250

0xc0170, 2019-03-05:
> <built-in>:388:10: fatal error: 'mbed_config.h' file not found
#include "./mbed_config.h"

@ARMmbed/mbed-os-tools This might indicate a bug in our code, however `mbed_config.h` has been working with make or others. Please review, this is blocking yours 2 PRs for 5.12rc1 (besides others)

alekla01, 2019-03-05:
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/__init__.py#L853
**failure:**
``
self.config_file = join(self.build_dir, self.MBED_CONFIG_FILE_NAME) // './mbed_config.h'
``
**success:**
``
import os
self.config_file = os.path.abspath(join(self.build_dir, self.MBED_CONFIG_FILE_NAME)) // '/builds/ws/mbed-os-ci_exporter-make_armc6_3/examples/mbed-os-example-mesh-minimal/mbed_config.h'
``

alekla01, 2019-03-05:
or join(os.path.abspath(self.build_dir), self.MBED_CONFIG_FILE_NAME)

theotherjimmy, 2019-03-05:
@alekla01 Using an abspath will kill all exports from the online compiler.

theotherjimmy, 2019-03-05:
I have root-caused the issue here: Adding the include paths to the assembler command line is the problem. Specifically, the Makefile we export is a little sophisticated, and runs the compiler from within the build directory. The assembly preprocessing does not take this into account, and creates include paths that won't exist in the build directory.

alekla01, 2019-03-05:
-> '..' instead of abspath, ok

0xc0170, 2019-03-05:
> I have root-caused the issue here: Adding the include paths to the assembler command line is the problem. Specifically, the Makefile we export is a little sophisticated, and runs the compiler from [...]

How to fix this? 5.12 rc1 is blocked on this issue.

bridadan, 2019-03-05:
I'm taking a look at this now.

PR #9946 (2019-03-06): Fix include paths for the assembler for makefiles
### Description

Should fix #9938.

Seems to work fine with ARMC5, ARMC6, and GCC_ARM.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@0xc0170 - I tried my darndest to get git to ignore the whitespace changes, but I couldn't figure it out. If you feel really strongly about it let me know.

### Release Notes
closed 2019-03-06 by cmonr