ARMmbed/mbed-os #9945: STM32F746 Discovery Can Bus Problem

Zborex, 2019-03-05:
### Description
Hello,
I have a problem on Disco F746 with startup while Can bus is active.
I get error Code on stlink terminal:
can ESR  0x0100.0010 + timeout status 0

Device freezes and is inoperable. Only way to start the device is to power it up and then turn on the CanBUS. I am checking this with Kvaser software, but in real environment the freeze occurs aswell, more rarely. If I choose can mode to Silent, I have no problem on startup, but unable to send messages.
Code:
```

CAN can1(PB_8,PB_9);

void canConfig(int canspeed)
{ 
   can1.mode(can1.Normal);
   can1.frequency(canspeed*1000);
   can1.attach(&CanRx,can1.RxIrq);
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-14:
cc @ARMmbed/team-st-mcd Can you please review?

MubeenHCLite, 2021-05-03:
@jeromecoutant 

The code I have used as below:
main.cpp

```
#include "mbed.h"

CAN can1(PB_8,PB_9);
DigitalOut led1(LED1);
DigitalOut led2(LED2);
DigitalOut led3(LED3);

void canConfig(int canspeed)
{ 
   can1.mode(can1.Normal);
   can1.frequency(canspeed*1000);
   //can1.attach(&CanRx,can1.RxIrq);
}

int counter = 0xABCD;
 
void send() {
    printf("send()\r\n");
        if(can1.write(CANMessage(1102, (char *)&counter, 2))) {
        printf("loop send()\r\n");
        counter++;
        printf("Message sent: %d\r\n", counter);
    }
}
 
int main() {
    CANMessage msg;
    while(1) {
        //send();
        //wait_us(500000);
        printf("loop wait()\r\n");
        if(can1.read(msg)) {
            printf("Message received: %x%x from %d\r\n", msg.data[0], msg.data[1], msg.id);
            if(msg.id==1100)
                led2 = !led2;
            if(msg.id==1102)
                led3 = !led3;     
        }
        wait_us(500000);
    }
}
```

mbed-os/cmsis/device/rtos/source/mbed_boot.c: 
```
#include <stdlib.h>

#include "cmsis.h"
#include "hal/us_ticker_api.h"
#include "mbed_toolchain.h"
#include "mbed_boot.h"
#include "mbed_error.h"
#include "mbed_mpu_mgmt.h"

int main(void);
void canConfig(int canspeed);
static void mbed_cpy_nvic(void);

/* Stack limits */
unsigned char *mbed_stack_isr_start = 0;
uint32_t mbed_stack_isr_size = 0;

void mbed_init(void)
{
#ifdef MBED_DEBUG
    // Configs to make debugging easier
#ifdef SCnSCB_ACTLR_DISDEFWBUF_Msk
    // Disable write buffer to make BusFaults (eg write to ROM via NULL pointer) precise.
    // Possible on Cortex-M3 and M4, not on M0, M7 or M33.
    // Would be less necessary if ROM was write-protected in MPU to give a
    // precise MemManage exception.
    SCnSCB->ACTLR |= SCnSCB_ACTLR_DISDEFWBUF_Msk;
#endif
#endif
    mbed_mpu_manager_init();
    mbed_cpy_nvic();
    mbed_sdk_init();
#if DEVICE_USTICKER && MBED_CONF_TARGET_INIT_US_TICKER_AT_BOOT
    us_ticker_init();
#endif
    mbed_rtos_init();
}

void mbed_start(void)
{
    mbed_rtos_init_singleton_mutex();
    mbed_tfm_init();
    mbed_toolchain_init();
    mbed_main();
    mbed_error_initialize();
    main();
}

MBED_WEAK void mbed_sdk_init(void)
{
    // Nothing by default
}

MBED_WEAK void software_init_hook_rtos()
{
    // Nothing by default
}

MBED_WEAK void mbed_tfm_init(void)
{
    // Nothing by default
}

MBED_WEAK void mbed_main(void)
{
    canConfig(1000);
}

static void mbed_cpy_nvic(void)
{
    /* If vector address in RAM is defined, copy and switch to dynamic vectors. Exceptions for M0 which doesn't have
    VTOR register and for A9 for which CMSIS doesn't define NVIC_SetVector; in both cases target code is
    responsible for correctly handling the vectors.
    */
#if !defined(__CORTEX_M0) && !defined(__CORTEX_A9)
#ifdef NVIC_RAM_VECTOR_ADDRESS
    uint32_t *old_vectors = (uint32_t *)SCB->VTOR;
    uint32_t *vectors = (uint32_t *)NVIC_RAM_VECTOR_ADDRESS;
    for (int i = 0; i < NVIC_NUM_VECTORS; i++) {
        vectors[i] = old_vectors[i];
    }
    SCB->VTOR = (uint32_t)NVIC_RAM_VECTOR_ADDRESS;
#endif /* NVIC_RAM_VECTOR_ADDRESS */
#endif /* !defined(__CORTEX_M0) && !defined(__CORTEX_A9) */
}
```

MubeenHCLite, 2021-05-18:
@Zborex 
We are unable to reproduce the issue. 
Could you add more details like which version of mbed library were you using?
Your CanRx Implementation.
Do let know what speed you are running you CAN module at.

adbridge, 2021-06-07:
@Zborex we are unable to reproduce your issue and without further information as per the request 3 weeks ago, we cannot progress further. We will close this for now, please re-open if this is still an issue and if so, provide the required information. Thanks.

closed 2021-06-07 by adbridge