ARMmbed/mbed-os #9949: Bootloader Merging Regions Error

Sim-jaeil, 2019-03-06:
### Description
Hi. I am using the MBED boot loader function. The first attached picture below is MBED-OS-5.11.0, and it works without error. The second attached picture is a picture of MBED-OS-5.11.1 version, and all subsequent versions will behave similarly. The problem is that the "Space used after regions merged" subaddress is different from the APP start address you set. Why is that? What changed the boot loader method? I'm sorry I can not speak English well. Thank you.
[image]
[image]

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

pawelm87, 2019-03-21:
I have this same problem. After compilate bootloader I got two files bin:

- mbed-os-example-bootloader.bin - 121 KB
- mbed-os-example-bootloader_application.bin - 121 KB

Two file have this same size and (binary) looks that same.
When I used "mbed-os-example-bootloader_application.bin" to build application it doesn't work. Bootloader load app from SD, flashed memory, print message "Starting application" and stop. App don't start. The tutorial on the [github](https://github.com/armmbed/mbed-os-example-bootloader) website states that there should be two different files for the bootloader to work.

When I change folder "tools" (old version mbed-os probably 5.10.4) in mbed-os everything work fine, I got two file with difference size. App builded from file padded bootloader - start after flashing memory. 

> Compiler GCC_ARM
> STM32L476 - I changed mbed_app.json - bootloader works on mbed-os 5.10.4
> mbed-os 5.11.5 - imported program in 20.03.2019 from git
> program:
> - mbed-os-example-bootloader 
> - mbed-os-example-blinky
>
> in Mbed Studio change folder "tools" don't work 

Can you fix this bug or told what happens ??

cmonr, 2019-03-26:
@ARMmbed/mbed-os-tools

mfatiga, 2019-04-03:
What is the status on this? This seems to be a big issue, not being able to build a bootloader at all. Is there any temporary fix that could be done until this is solved?

theotherjimmy, 2019-04-03:
> not being able to build a bootloader at all

That's not what this issue is about. This issue is about missing padding after a bootloader is built. If you can't build the bootloader, please open a different issue.

mfatiga, 2019-04-03:
@theotherjimmy yes, I perfectly understand what the issue is about. Bu the final bootloader image is not correct because it is missing the padding. Sorry if I expressed myself incorrectly. I have solved this in our environment in our build scripts by validating the bootloader image size with the restrict_size value in mbed_app.json and padding it if necessary. We have projects with integrated bootloader source and main application source and this issue broke our build because the main app does not start. 
Shouldn't this kind of bug be detected by tests?

madchutney, 2019-07-25:
This looks like it was caused by PR https://github.com/ARMmbed/mbed-os/pull/8097.

boraozgen, 2020-08-25:
This issue is present in `mbed-os-6.2.0`. Is this broken again or never fixed?

mfatiga, 2020-08-25:
@boraozgen Yes, looks like this is still not fixed, I am using a custom python script that is validating the bootloader image size with the restrict_size value in mbed_app.json and padding it if necessary using intelhex.

0xc0170, 2021-09-09:
For anyone interested, there was a fix referenced above. As the old tools were frozen, I'll close this issue as won't fix.

closed 2021-09-09 by 0xc0170