ARMmbed/mbed-os #9950: BLE App is getting crashed after flash it using greentea

UrmilaKL, 2019-03-06:
Board: DISCO_L475VG_IOT01A
mbedls: 1.7.3
mbed os: 5.11.5
mbed cli: 1.8.3

BLE App is getting crashed after flash it using greentea 
Hard fault error observed soon after flashing it on to DISCO board,
I have merged both heartrate and thermometer example into one example app and trying to flash it 

Please find the attached main.cpp

Thanks 
Urmila Bhat

UrmilaKL, 2019-03-06:
#include <events/mbed_events.h>
#include <mbed.h>
#include "ble/BLE.h"
#include "ble/Gap.h"
#include "ble/services/HeartRateService.h"
#include "ble/services/HealthThermometerService.h"

#define USE_BOARD_TEMP_SENSOR

#ifdef USE_BOARD_TEMP_SENSOR
#include "stm32l475e_iot01_tsensor.h"
#endif

DigitalOut led1(LED1, 1);

const static char     DEVICE_NAME[] = "MyExampleApp";
static const uint16_t uuid16_list[] = {GattService::UUID_HEART_RATE_SERVICE, GattService:: UUID_HEALTH_THERMOMETER_SERVICE};

//size_t malloc_size = 16 * 1024;

static uint8_t hrmCounter = 100; // init HRM to 100bps
static HeartRateService *hrServicePtr;
static float currentTemperature   = 39.6;
static HealthThermometerService *thermometerServicePtr;

static EventQueue eventQueue(/* event count */ 16 * EVENTS_EVENT_SIZE);

void disconnectionCallback(const Gap::DisconnectionCallbackParams_t *params)
{
    BLE::Instance().gap().startAdvertising(); // restart advertising
}

void updateSensorValue() {
    // Do blocking calls or whatever is necessary for sensor polling.
    // In our case, we simply update the HRM measurement.
    hrmCounter++;

    //  100 <= HRM bps <=175
    if (hrmCounter == 175) {
        hrmCounter = 100;
    }
#ifdef USE_BOARD_TEMP_SENSOR
    currentTemperature = BSP_TSENSOR_ReadTemp();
#else
    currentTemperature = (currentTemperature + 0.1 > 43.0) ? 39.6 : currentTemperature + 0.1;
#endif
    thermometerServicePtr->updateTemperature(currentTemperature);
    hrServicePtr->updateHeartRate(hrmCounter);
    printf("\nHeartRate Measured value is %d \n ", hrmCounter);
    printf("\nTemprature Value %f\n", currentTemperature);
}

void periodicCallback(void)
{
    led1 = !led1; /* Do blinky on LED1 while we're waiting for BLE events */

    if (BLE::Instance().getGapState().connected) {
        eventQueue.call(updateSensorValue);
    }
}

void onBleInitError(BLE &ble, ble_error_t error)
{
    (void)ble;
    (void)error;
   /* Initialization error handling should go here */
}

void bleInitComplete(BLE::InitializationCompleteCallbackContext *params)
{
    BLE&        ble   = params->ble;
    ble_error_t error = params->error;

    if (error != BLE_ERROR_NONE) {
        onBleInitError(ble, error);
        return;
    }

    if (ble.getInstanceID() != BLE::DEFAULT_INSTANCE) {
        return;
    }

    ble.gap().onDisconnection(disconnectionCallback);

    /* Setup primary service. */
    hrServicePtr = new HeartRateService(ble, hrmCounter, HeartRateService::LOCATION_FINGER);
    thermometerServicePtr = new HealthThermometerService(ble, currentTemperature, HealthThermometerService::LOCATION_EAR);
    

    /* Setup advertising. */
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::BREDR_NOT_SUPPORTED | GapAdvertisingData::LE_GENERAL_DISCOVERABLE);
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LIST_16BIT_SERVICE_IDS, (uint8_t *)uuid16_list, sizeof(uuid16_list));
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::GENERIC_HEART_RATE_SENSOR);
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::THERMOMETER_EAR);
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LOCAL_NAME, (uint8_t *)DEVICE_NAME, sizeof(DEVICE_NAME));
    ble.gap().setAdvertisingType(GapAdvertisingParams::ADV_CONNECTABLE_UNDIRECTED);
    ble.gap().setAdvertisingInterval(1000); /* 1000ms */
    ble.gap().startAdvertising();  
}

void scheduleBleEventsProcessing(BLE::OnEventsToProcessCallbackContext* context) {
    BLE &ble = BLE::Instance();
    eventQueue.call(Callback<void()>(&ble, &BLE::processEvents));
}

int main()
{
    #ifdef USE_BOARD_TEMP_SENSOR
    BSP_TSENSOR_Init();
    #endif
	
    printf("My Example Application Initialized !\n");
    eventQueue.call_every(1000, periodicCallback);

    BLE &ble = BLE::Instance();
    ble.onEventsToProcess(scheduleBleEventsProcessing);
    ble.init(bleInitComplete);

    eventQueue.dispatch_forever();

    return 0;
}

desmond-blue, 2019-03-12:
Hi @UrmilaKL,

After looking you main.cpp, it's not a greentea architecture, what test you would like to do with green tea?

However it seems like a good BLE application, how about using -f to flash the firmware like below and see if the crash still happens.
`mbed compile -t [Your compiler] -m DISCO_L475VG_IOT01A -f`

Could you provide the crash log?

UrmilaKL, 2019-03-12:
Hi @Desmond,

Any kind of testcase using greentea/icetea which should be able to communicate with BLE API
is fine for me 
I will take up the example to explore more for my project
Thanks in advance
Urmila Bhat

UrmilaKL, 2019-03-12:
Hi @Desmond,

It would be better if you can provide some BLE test cases to set the ADV API parameters 

Thanks

desmond-blue, 2019-07-30:
Hi @UrmilaKL, do you still need help on this issue?

desmond-blue, 2019-08-26:
@0xc0170 I guess we can close it since no feedback from reporter for almost one month?

closed 2019-09-18 by ciarmcom