ARMmbed/mbed-os #9951: LoRaWAN stack halts when encountering 'RX1 Demodulation ongoing, skip RX2 window opening'

janjongboom, 2019-03-06:
### Description

Not sure if this is a valid issue as it only shows up when setting `max-sys-rx-error`, but when the LoRaWAN stack encounters:

```
[INFO][LMAC]: RX1 Demodulation ongoing, skip RX2 window opening
```

Everything halts. I'd expect some error to go back to the application, but now the state machine doesn't continue at all. It also doesn't continue demodulating the RX1 message.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

hasnainvirk, 2019-03-27:
@janjongboom this can be closed I guess. The issue was related to MCU clock speed as we had discussed,

janjongboom, 2019-03-27:
@hasnainvirk I'm not sure if I agree that this can be closed. This is a race condition where the stack ends up in unrecoverable state. That should never be the case.

hasnainvirk, 2019-03-27:
@janjongboom I beg to differ. For example, 
SF=12, BW = 125kHz, CR= 4/5, Payload size = 20 bytes, Preamble size to have a lock = 6 (+ 4.25 what chip adds) will give you TOA 1089.54 ms. That means it is guaranteed that  when there will be time to open RX2 window, you would still be demodulating RX1. 
The problem you were seeing was reproducible and after the clock speed fix, it went away. However, there was a bug which would result in premature closure of RX2 window leaving RX2 barely usable. It was fixed here https://github.com/ARMmbed/mbed-os/pull/10196

janjongboom, 2019-03-27:
@hasnainvirk OK I'll trust you on it!

closed 2019-03-27 by janjongboom

Eric-A-Marks, 2021-01-26:
@janjongboom or @hasnainvirk , do either of you remember what the clock speed fix was?  I'm investigating what appears to be some race conditions in the lorawan stack and any insight would be appreciated.

hasnainvirk, 2021-01-30:
@Eric-A-Marks This was reported while playing around SAML/R32 platform if I rememnber correctly. :) That board was being clocked at 4MHz rather than 40 MHz.