ARMmbed/mbed-os #9958: nRF52 VDD measurement

marcemmers, 2019-03-06:
### Description

I am currently developing on a custom nRF52 target and I need to measure the VDD for some analog calculations. The nRF52 targets at least have an option for internal measurement of the VDD and VDDH but no option in mbed to select that channel.

Currently the 8 possible AnalogIn channels are mapped to the 8 channels of the SAADC but only one channel is activated at a time. A possibility to solve this is to add the VDD and VDDH channels as pins that can only be used with AnalogIn. The analogin_api should only use channel 0 of the SAADC and configure the correct positive input before the measurement based on the selected pin.

We are currently using OS version 5.11.1 but it's not changed in the newer releases.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

TacoGrandeTX, 2019-03-29:
@marcemmers Sorry for the long delay in responding to this.  You are right - we are not allowing for VDD to be an option for sampling, so thanks for pointing out this deficiency.  Ultimately I think you are concerned about setting the input pin selection, ```CH[x].PSELP``` or ```CH[x].PSELN```, to VDD.    

If so, I think there's a much simpler fix than what you are proposing.  All that we need to do is expand the definition of the nrf_saadc_channel_t enum to include VDD with value 8 to match the Nordic Product Spec:  

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/PeripheralPins.c#L41-L50

And you can then use this value as required.  If we're off base can you clarify?

marcemmers, 2019-04-01:
Setting PSELP to VDD is not really an issue. Adding VDD and VDDH to the nrf_saadc_channel_t enum is indeed part of the solution.

Why I think its more difficult than what you propose is that the SAADC only has 8 channel slots in the peripheral, at least in the nRF52840. The 8 analog inputs that are available now are mapped to those configuration slots so there is no room for the VDD and VDDH channels. See:
https://github.com/ARMmbed/mbed-os/blob/1b4f6ae3c4ef0633d399567cd346322a461e42d2/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c#L119

I have fixed it for us internally, but only for the 840: https://github.com/marcemmers/mbed-os/commit/26cbda6811aac7466d30574a735e42411a7a4475

desmond-blue, 2019-07-31:
Hi @marcemmers , since you already have a patch, do you need further help on this issue? Thanks.

marcemmers, 2019-08-01:
Hi @desmond-blue, we have patched it on our own fork but its not yet fixed in mbed. See my comment from april first.

As i've said, there are to many available inputs for the ADC to make use of and configure the internal peripheral channels. My proposal is to use one single ADC channel and configure that channel each time a measurement needs to be made. This way the nrf_saadc_channel_t enum could be expanded to include the VDD and VDDH inputs.

closed 2019-10-31 by ciarmcom