ARMmbed/mbed-os #9961: nRF52840 - InterruptIn

DBS06, 2019-03-06:
### Description
According to "nRF52840 Product Specification v1.0" 8 InterruptIn's are available,  I tried to use 7 InterruptIn's but if I try to initialize more than 4 mbed quits with the following error:

> ++ MbedOS Error Info ++
> Error Status: 0x80FF0100 Code: 256 Module: 255
> Error Message: Fatal Run-time error
> Location: 0x3266B
> Error Value: 0x0
> Current Thread: main  Id: 0x2001612C Entry: 0x4FDC7 StackSize: 0x4000 StackMem: 0x200120F8 SP: 0x20015EA0 
> For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
> -- MbedOS Error Info --
> nrf failure at p:\ng-master-app\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_14_2\drivers_nrf\gpiote\nrf_drv_gpiote.c:612

The asseration at `nrf_drv_gpiote.c:612` does the following:
`ASSERT(pin_in_use_by_gpiote(pin));`

Why does this happen and how can I avoid it?

KR

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

DBS06, 2019-03-07:
I think I found the solution.
In `"mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_NRF52\TARGET_MCU_NRF52840\config\sdk_config.h"   ` the define `GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS 4` should be changed.

This can be done inside the custom targets with
`"macros_add": ["GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8"]`

Hopefully this helps

0xc0170, 2019-03-08:
> I think I found the solution.

Can this be closed ?

j3hill, 2019-03-08:
Yes. We can close this.

DBS06, 2019-03-08:
Btw the "Interrupt's" are actually "Events" so it is possible to get more than 8.

closed 2019-03-08 by DBS06