ARMmbed/mbed-os #9968: Unable to debug/run CY8CKIT_062_WIFI_BT using Eclipse

linlingao, 2019-03-06:
### Description

Steps to reproduce:
(1) mbed import mbed-os-example-wifi
(2) cd mbed-os-example-wifi
(3) Create mbed_app.json if it's not at the root directory. Then add your SSID and password.
(4) mbed compile -m CY8CKIT_062_WIFI_BT -t gcc_arm --profile debug
(5) Copy generated mbed-os-example-wifi.hex to the enumerated DAPLINK drive
(6) Press RST button after the image has been copied over
(7) Open serial terminal and verify the program runs to completion. You should see the board connects to the access point and MAC/IP address etc. are acquired. Something like:

Connecting to Netgear_N300...
Success

MAC: a0:c9:a0:46:0f:97
IP: 10.118.122.91
Netmask: 255.255.254.0
Gateway: 10.118.122.1
RSSI: -35

Done
(8) mbed export -m CY8CKIT_062_WIFI_BT -i gcc_arm --profile debug
(9) Import it to Eclipse. Then Build Project
(10) Create a debug configuration using pyOCD. Make sure Board is seen and selected. Set breakpoint at main.
(11) Click Debug
(12) Expect the debugger to halt the MCU at main. On this board, the debugger is not stopped. When I pause the debugger, I see PC at 0x8000378.
[image]
It appears the elf might be missing content at 0x10000000.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

maclobdell, 2019-03-07:
cc @ARMmbed/team-cypress

AndriiLishchynskyi, 2019-03-07:
Hi @linlingao,

I assume you didn't take into account that CY8CKIT_062_WIFI_BT is dual-core device and created launch configuration is intended to debug CM0+ core. Make sure to specify second core to debug:
[image]

Also you may get valid launch configurations by running:
mbed export -i eclipse_gcc_arm -m CY8CKIT_062_WIFI_BT --profile debug

linlingao, 2019-03-07:
@AndriiLishchynskyi Thank you so much! After changing the debug configuration, I'm able to debug.
It would be very helpful if these steps are documented on the board page on os.mbed.com.

closed 2019-03-07 by linlingao