ARMmbed/mbed-os #9969: Network test fails on CY8CKIT_062_WIFI_BT and CY8CPROTO_062_4343W

maclobdell, 2019-03-07:
### Description

I'm running mbed os network tests on CY8CKIT_062_WIFI_BT and CY8CPROTO_062_4343W. 

Version: Mbed OS 5.15.5

Toolchains: ARM and GCC_ARM

Test commands used: 

mbed test -t GCC_ARM -m CY8CPROTO_062_4343W -n tests-network-*
mbed test -t GCC_ARM -m CY8CKIT_062_WIFI_BT -n tests-network-*

Test Setup: 
I have a WiFi access point in the close vicinity with the following credentials.   It has a upstream connection to the internet.  

           SSID = mbedEnabled
           password = validation
           security = WPA_WPA2

Configuration: 
I have a mbed_app.json file in the root directory with the following settings.

```
{
    "config": {
        "wifi-secure-ssid": {
            "help": "WiFi SSID for WPA2 secured network",
            "value": "\"test-network\""
        },
        "wifi-unsecure-ssid": {
            "help": "WiFi SSID for unsecure netwrok",
            "value": "\"unsecure-test-net\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"password\""
        },
        "wifi-secure-protocol": {
            "help": "WiFi security protocol, valid values are WEP, WPA, WPA2, WPA_WPA2",
            "value": "\"WPA2\""
        },
        "wifi-ch-secure": {
            "help": "Channel number of secure SSID",
            "value": 6
        },
        "wifi-ch-unsecure": {
            "help": "Channel number of unsecure SSID",
            "value": 6
        },
        "ap-mac-secure": {
            "help": "BSSID of secure AP in form of AA:BB:CC:DD:EE:FF",
            "value": "\"58:8b:f3:99:f2:9c\""
        },
        "ap-mac-unsecure": {
            "help": "BSSID of unsecure AP in form of \"AA:BB:CC:DD:EE:FF\"",
            "value": "\"58:8b:f3:99:c2:08\""
        },
        "max-scan-size": {
            "help": "How many networks may appear in Wifi scan result",
            "value": 30
        },
        "echo-server-addr" : {
            "help" : "IP address of echo server",
            "value" : "\"echo.mbedcloudtesting.com\""
        },
        "echo-server-port" : {
            "help" : "Port of echo server",
            "value" : "7"
        },
        "echo-server-port-tls" : {
            "help" : "Port of echo server TLS",
            "value" : "2007"
        }
    },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "WIFI",
            "nsapi.default-wifi-ssid": "\"mbedEnabled\"",
            "nsapi.default-wifi-password": "\"validation\"",
            "nsapi.default-wifi-security": "WPA_WPA2"
        }
    }
}
````

Results:
```
| target                      | platform_name       | test suite              | test case                          | passed | failed | result  | elapsed_time (sec) |
|-----------------------------|---------------------|-------------------------|------------------------------------|--------|--------|---------|--------------------|
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-interface | NETWORKINTERFACE_CONN_DISC_REPEAT  | 1      | 0      | OK      | 17.27              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-interface | NETWORKINTERFACE_STATUS            | 1      | 0      | OK      | 24.82              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-interface | NETWORKINTERFACE_STATUS_GET        | 1      | 0      | OK      | 17.25              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-interface | NETWORKINTERFACE_STATUS_NONBLOCK   | 1      | 0      | OK      | 17.29              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-interface | WIFI-CONNECT-DISCONNECT-REPEAT     | 0      | 20     | FAIL    | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT                       | 0      | 6      | FAIL    | 107.79             |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-DISCONNECT-REPEAT     | 0      | 0      | ERROR   | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-NOCREDENTIALS         | 1      | 0      | OK      | 0.22               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-PARAMS-CHANNEL        | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-PARAMS-CHANNEL-FAIL   | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-PARAMS-NULL           | 1      | 0      | OK      | 0.04               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-PARAMS-VALID-SECURE   | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-PARAMS-VALID-UNSECURE | 0      | 4      | FAIL    | 71.91              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-SECURE                | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONNECT-SECURE-FAIL           | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-CONSTRUCTOR                   | 1      | 0      | OK      | 0.05               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-GET-RSSI                      | 0      | 2      | FAIL    | 35.82              |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-SCAN                          | 0      | 1      | FAIL    | 0.42               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-SCAN-NULL                     | 1      | 0      | OK      | 0.39               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-SET-CHANNEL                   | 1      | 0      | OK      | 0.08               |
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-network-wifi      | WIFI-SET-CREDENTIAL                | 1      | 0      | OK      | 0.05               |
mbedgt: test case results: 5 FAIL / 5 SKIPPED / 10 OK / 1 ERROR

```

Possible test setup issues: 
    As I type this, I realize that I haven't checked that the connection to the internet is working on other boards.  I will check that asap.  

Please help!  

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

maclobdell, 2019-03-07:
cc @SeppoTakalo - any suggestions?

SeppoTakalo, 2019-03-08:
@maclobdell First, correct the WiFi SSID configurations

```
        "wifi-secure-ssid": {
            "help": "WiFi SSID for WPA2 secured network",
            "value": "\"test-network\""
        },
        "wifi-unsecure-ssid": {
            "help": "WiFi SSID for unsecure netwrok",
            "value": "\"unsecure-test-net\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"password\""
        },
        "wifi-secure-protocol": {
            "help": "WiFi security protocol, valid values are WEP, WPA, WPA2, WPA_WPA2",
            "value": "\"WPA2\""
        },
```
Those values are not matching what you said your test network is.

Also, please note that for WiFi tests, you need two SSIDs. One for secure, and one for unsecure.
See: https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md#wi-fi-test-environment

maclobdell, 2019-03-08:
thanks.  I thought everything necessary was correct in the targets override section.  I also thought that the insecure access point was optional.  My apologies.  I'll correct the test setup.

maclobdell, 2019-03-08:
why do the tests not use the nsapi.default values?

            "nsapi.default-wifi-ssid": "\"mbedEnabled\"",
            "nsapi.default-wifi-password": "\"validation\"",
            "nsapi.default-wifi-security": "WPA_WPA2"

maclobdell, 2019-03-08:
I just retested with the suggestions.  Everything passes.  The issue is resolved.  My mistake.  Thanks!

closed 2019-03-08 by maclobdell