ARMmbed/mbed-os #9970: "latest" tag is still on 5.11.4 commit

jackyhuang85, 2019-03-07:
### Description

* Target: Any
* toolchain: GCC_ARM
* tools: mbed-cli v1.9.1

#### Reproduce
1. `mbed new <project name>`
2. It appears `[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"`
3. `mbed compile -t GCC_ARM -m K66F` -> 5.11.4 py3 bug happened

1. `mbed import <example-blinky>`
2. `[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #51d55508e840`
3. `mbed compile -t GCC_ARM -m K66F` -> run successfully

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

As mentioned in https://github.com/ARMmbed/mbed-cli/issues/867#issuecomment-470384282 ,
the "latest" tag is still at `#ecb3c8c8`, the version 5.11.4, it causes mbed-cli `mbed new` pulls the older version mbed-os library.

0xc0170, 2019-03-07:
Hi,

thanks for the report, we will check. From the view on Github, looks like https://github.com/ARMmbed/mbed-os/releases/tag/latest is still pointing to the older one.

cc @ARMmbed/mbed-os-maintainers

0xc0170, 2019-03-07:
Fixed https://github.com/ARMmbed/mbed-os/releases/tag/latest

Closing as resolved

closed 2019-03-07 by 0xc0170