ARMmbed/mbed-os #9974: LPC54114: iar exporter fails with fpu keyerror

0xc0170, 2019-03-07:
### Description

Target LPC54114 fails for IAR exporter

```

    Scan: mbed-os-example-blinky
Traceback (most recent call last):
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\project.py", line 407, in <module>
    main()
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\project.py", line 399, in main
    ignore=options.ignore
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\project.py", line 171, in export
    ignore=ignore
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\export\__init__.py", line 284, in export_project
    macros=macros)
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\export\__init__.py", line 157, in generate_project_files
    exporter.generate()
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\export\iar\__init__.py", line 126, in generate
    debugger = DeviceCMSIS(self.target).debug.replace('-','').upper()
  File "C:\builds\ws\mbed-os-ci_exporter-iar8@2\examples\mbed-os-example-blinky\mbed-os\tools\export\cmsis\__init__.py", line 50, in __init__
    self.dfpu = target_info['processor']['Asymmetric']['fpu']
KeyError: 'fpu' 
```

Log can be found here http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/144/exporter/FAIL/iar/LPC54114/

Found on 5.12.0-rc1 nightly as well as on master

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

@ARMmbed/team-nxp @ARMmbed/mbed-os-tools

theotherjimmy, 2019-03-07:
Good catch; good report. I'll look at it shortly

theotherjimmy, 2019-03-07:
I cannot reproduce this.

PR #9994 (2019-03-14): Correct FPU settings traceback found by IAR
### Description

Fixes #9974

The error was that the Asymmetric CPUs were assumed to have the same
structure as the Symmetric CPUs. This is clearly false.

This PR changes the FPU detection of Asymmetric CPUs to find the correct
core and use it's cpu settings.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
commit 35479e81 (2019-03-14): Correct FPU settings traceback found by IAR

Fixes #9974

The error was that the Asymmetric CPUs were assumed to have the same
structure as the Symmetric CPUs. This is clearly false.

This PR changes the FPU detection of Asymmetric CPUs to find the correct
core and use it's cpu settings.

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
commit e1a2c0cd (2019-03-16): Correct FPU settings traceback found by IAR

Fixes #9974

The error was that the Asymmetric CPUs were assumed to have the same
structure as the Symmetric CPUs. This is clearly false.

This PR changes the FPU detection of Asymmetric CPUs to find the correct
core and use it's cpu settings.

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
commit a197ebca (2019-03-19): Correct FPU settings traceback found by IAR

Fixes #9974

The error was that the Asymmetric CPUs were assumed to have the same
structure as the Symmetric CPUs. This is clearly false.

This PR changes the FPU detection of Asymmetric CPUs to find the correct
core and use it's cpu settings.

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2019-03-14 by 0xc0170