ARMmbed/mbed-os #9975: ublox: UBLOX_EVK_ODIN_W2 hard faults for wifi test (armc6)

0xc0170, 2019-03-07:
### Description

Logs can  be found Test logs: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/mbed-os-5.12/artifacts/mbed-os-5.12/2/greentea-test/1406/ERROR/UBLOX_EVK_ODIN_W2/ARMC6/

This is in nightly. Wifi seems to be hard faulting for ARMC6.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cc @ARMmbed/team-ublox 

@ARMmbed/mbed-os-ipcore Known issue or new one?

michalpasztamobica, 2019-03-07:
Thanks, Martin. Yes, we've also noticed and we are already working on it in ONME-4220.

0xc0170, 2019-03-07:
Great! Once this is cloned to our jira, please link these 2 issues and fix should close this one.

michalpasztamobica, 2019-03-07:
My findings are that [this commit](https://github.com/ARMmbed/mbed-os/commit/30f18b67e551f7cae83c843cddefdd2735e6273b) introduced the issue (switching to ARMC6) and that the hard fault takes place when [this line](https://github.com/ARMmbed/mbed-os/blob/aedc1e10ead5593086850524ced46ce60541e33b/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp#L314) is executed.

The hard fault message is:

```
[1551962056.94][CONN][RXD] ++ MbedOS Error Info ++
[1551962056.94][CONN][RXD] Error Status: 0x80FF013D Code: 317 Module: 255
[1551962056.94][CONN][RXD] Error Message: Fault exception
[1551962056.96][CONN][RXD] Location: 0x801B64D
[1551962056.96][CONN][RXD] Error Value: 0x800D8BE
[1551962056.96][CONN][RXD] Current Thread: application_unnamed_thread  Id: 0x2000742C Entry: 0x8010FD1 StackSize: 0x1000 StackMem: 0x20012940 SP: 0x2002FF18
[1551962056.98][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=UBLOX_EVK_ODIN_W2
[1551962056.98][CONN][RXD] -- MbedOS Error Info --
[1551962056.98][CONN][RXD]
[1551962056.98][CONN][RXD] = System will be rebooted due to a fatal error =
[1551962056.99][CONN][WRN] UnicodeDecodeError encountered!
```
@ARMmbed/team-ublox , would you please take a look at this issue?

0xc0170, 2019-03-08:
> My findings are that this commit introduced the issue (switching to ARMC6) and that the hard fault takes place when this line is executed.

> @ARMmbed/team-ublox , would you please take a look at this issue?

Is that correct the commit above is causing this hard fault? Any update?

cc @screamerbg

0xc0170, 2019-06-07:
@ARMmbed/team-ublox Just want to confirm, the next driver update should have this resolved?

aqib-ublox, 2019-06-11:
@0xc0170  here is internal jira for ARMC6 https://jira.u-blox.net/browse/TE_OM2-115. point is can u device a mechanism how to debug a ARMC6 (ODIN_w2) binary using eclipse. It would be good if u can have detail procedure.

0xc0170, 2019-06-11:
> @0xc0170 here is internal jira for ARMC6 https://jira.u-blox.net/browse/TE_OM2-115. point is can u device a mechanism how to debug a ARMC6 (ODIN_w2) binary using eclipse. It would be good if u can have detail procedure.

would you use mbed studio? I'll contact you over email to get more details

0xc0170, 2019-06-11:
This is the official documentation we have : https://os.mbed.com/docs/mbed-os/v5.12/tutorials/eclipse.html 

Isn't this sufficient to keep you running debugging session ? It uses eclipse + openOCD (odin supported there, not yet in pyOCD), or ST provides gdb as well?)

0xc0170, 2019-06-11:
I also found private notebook describing how to set up openOCD with odin (it's almost 2 years old, might be still a good start): https://os.mbed.com/users/janjongboom/notebook/debugging-ublox_evk_odin_w2-with-openocd-and-visua/#

You should be able to debug armc6 application using gdb in eclipse. These 2 links should help you. Without much details where you are having issues, we might not be able to help further. 

It might be good to start a new issue to get this eclipse + armc6 question further - I would recommend asking on the forum, there are users who use eclipse daily.

SeppoTakalo, 2019-06-11:
https://github.com/mbedmicro/pyOCD/tree/master/pyocd/probe/stlink

There seems to be some development regarding ST-Link support for pyOCD.
Sounds like we need to have proper documentation / guide on our Handbook how to set up the debugger for most commom UIs and debuggers.

For example gdbservers:
* pyOCD + DAPlink
* pyOCD + ST-Link
* OpenOCD + ST-Link

Then UIs:
* command line GDB
* Eclipse + any of those gdbservers
* Arm studio
* Keil
* IAR

MarceloSalazar, 2019-06-19:
Depends on https://github.com/ARMmbed/mbed-os/issues/10814

aqib-ublox, 2019-06-28:
@0xc0170 I have tried mbed studio with 

1.  Go to https://os.mbed.com/studio/ download and install mbed studio with a double click
2.     File -> New Program... -> choose blinky from the example programs list
3.     Click build and wait (I can see that ARMC6 is used by default. Not sure how to change this to ARMC5 - is this important for you, @aqib-ublox ?)
4.     Click Debug

and able to see list of targets in target drop down menu. But it i clone project to local folder and followed below procedure

1. Clone blinky_led_example
2. compile project with --profile debug and export -i make_armc6
3. Open mbed studio and Open workspace
4. Now i cannot se any target in Target drop down menu?

Can u suggest i am doing anything wrong?

aqib-ublox, 2019-07-10:
have just fixed hard fault issue please see logs
[mbed-os-wifi-test.txt](https://github.com/ARMmbed/mbed-os/files/3376862/mbed-os-wifi-test.txt)

fix would be generated in next PR.

aqib-ublox, 2019-08-29:
https://github.com/ARMmbed/mbed-os/pull/11370 generated @0xc0170

closed 2019-10-25 by 0xc0170