ARMmbed/mbed-os #9983: lfs_emubd.c can be leaking resources

janjongboom, 2019-03-07:
### Description

In a number of functions in [lfs_emubd.c](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c) the functions are returning when f.e. `fread` fails. But this is not closing the file, thus leaking the resource.

Here are all of them (from cppcheck):

```
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:57]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:99]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:104]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:140]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:145]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:150]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:156]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:217]: (error) Resource leak: f
[features/storage/filesystem/littlefs/littlefs/emubd/lfs_emubd.c:233]: (error) Resource leak: f
```

@geky 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-08:
cc @ARMmbed/mbed-os-storage

Tharazi97, 2019-08-01:
Shouldn't this issue be placed on that repository?
[ARMmbed/littlefs](https://github.com/ARMmbed/littlefs)

davidsaada, 2019-08-01:
@geky

geky, 2019-08-05:
I'm going to be honest, this low priority right now. emubd is used primarily for testing, and when it errors, it errors.

If you're using emubd for real applications, it would likely be a better idea to map littlefs to a real disk/file. For example: https://github.com/ARMmbed/littlefs-fuse/blob/master/lfs_fuse_bd.c

I've moved this over here https://github.com/ARMmbed/littlefs/issues/260 so it's still tracked and can be taken off your guys's backlog.

0xc0170, 2019-10-25:
Closed as being tracked in the upstream repository where it should be fixed.

closed 2019-10-25 by 0xc0170