ARMmbed/mbed-os #9987: TARGET_TMPM46B/i2c_api.c does not check result of realloc

janjongboom, 2019-03-07:
### Description

In the TMPM46B target `gI2C_LTxData` is reallocated [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_TOSHIBA/TARGET_TMPM46B/i2c_api.c#L260) but the return value is not checked. If this fails it could lead to writing in uninitialized memory. 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-08:
cc @ARMmbed/team-toshiba

soramame21, 2019-03-11:
to @ganesh-ramachandran Could you check this issue please?

ganesh-ramachandran, 2019-03-12:
Hi @0xc0170, @soramame21,

We have fixed above issue if reallocation fails (PR #10048).

However, we are rewriting the I2C HAL driver code to support I2C slave and I2C asynchronous, in which we will remove the dynamic memory allocation mechanism.

closed 2021-05-12 by ciarmcom