ARMmbed/mbed-os #10006: CordioBLE stack : Sending Vendor specific command and receiving VSC complete event

harc-cypress, 2019-03-08:
### Description
Hi,
    In Cordio-stack i can see **HciVendorSpecificCmd** API  (in hci_api.h file).
    Application can use this API directly? or Is there any other way to send Vendor Specific command?
    Other thing is, in dmDevHciHandler function, **HCI_VENDOR_SPEC_CMD_CMPL_CBACK_EVT** event is not handle. So if app sends any vendor specific command, How app will get the status callback of sent command?
    Is it possible to send this event with status to the application?
Thanks,
Hardik

### Issue request type

    [x ] Question
    [x ] Enhancement
    [ ] Bug

pan-, 2019-03-08:
@harc-cypress Ideally that should be handled at the vendor driver level. 

We need to improve the driver interface so vendors can send and handle vendor specific commands. We also need to expose the vendor driver to end user so vendor specific capabilities can be use by applications.

harc-cypress, 2019-03-08:
@pan- Thanks for the quick response.
Do you already have any internal ticket for this improvement? 
How much approx time will it take to add this feature in cordio stack/driver?

pan-, 2019-03-08:
@harc-cypress Not yet, I will open an internal ticket today. Let's see where this goes with stakeholders. The HCI driver requires several improvements from an engineering perspective and we receive feature requests that we need to analyze.

ifyall, 2019-07-03:
@maclobdell Do we have an update on the priority of this improvement?

maclobdell, 2019-07-03:
@pan-   Please provide an update for the schedule of this feature to be implemented.  Cypress has middleware that depends on the implementation of #10361 or an alternative solution based on vendor specific commands.

ifyall, 2019-07-12:
Hello @pan-, is there any feedback on this or #10361? Thank you for your feedback.

pan-, 2019-07-12:
@donatieng

ifyall, 2019-09-23:
@donatieng Do you have any feedback on this? VSC is a critical capability for some vendor capabilities. Do you have a ticket in your backlog for this? Where is it in the priority queue?
Thank you,
Ian

donatieng, 2019-09-24:
Hi Ian,

I believe there were some discussions between Arm and Cypress on this issue (outside of GitHub). Let me find what the outcome was and I'll get back to you.

Don